prusa3d/Prusa-Firmware #261: 3.1.0 RCn  introduces stuttering on arcs and long moves

coldelectrons, 2017-10-31:
I have a MK2S with MMU, using Slic3r PE 1.37.1-prusa3d-linux64.

I was using firmware 3.0.12, which I changed to MMU version during my kit upgrade.

I thought I'd try this Linear Advance thing, so I have tried both firmwares 3.1.0 RC1 and RC2.

In all versions, the automatic movements of things like bed leveling are still smooth.

However, some longer moves and arcs (e.g. rapid from the end of the beginning nozzle purge to the first line of the part, or arcs in a skirt) that were mostly continuous motion in 3.0.12 have become jerky and stuttering, either with almost short pauses along a line, or sounding almost like grinding along an arc.  This behaviour happens with or without any K value.

First an image a test in 3.0.12, made in PLA that came with my Prusa:

[image]

 
Next, the same thing in 3.1.0-RC2:

[image]

It may  bit hard to tell in the photo - that's not just a single diagonal artifact in the center, it's an X; there are three base layers, and when the infill moves in each of them became long enough, there was a stutter at the midpoint.

Also note the skirt.  In 3.0.12, the arc motions are almost musical in the steppers, but in 3.1.0-RCn, they stutter so rapidly it almost sounds like the slip-clutch on a battery drill-driver.

Here is the file.  The only changes made between tests I have done is either K0, K200, or deleting that line entirely.
[Test_K000.txt](https://github.com/prusa3d/Prusa-Firmware/files/1432444/Test_K000.txt)

PavelSindler, 2017-11-02:
We are currently testing it. I have not seen any problem yet on RC2 MMU Rambo 1.0 and we are testing now on multiple MMU RC1 and RC2 Rambo 1.3.
Can you try to slow down your print to see if this helps? (by rotating the knob during printing)

PavelSindler, 2017-11-02:
So we finished our tests on multiple MMU printers and we also used the same filament to be as close as possible and we still can't see any issue.

Sebastianv650, 2017-11-02:
Maybe some acceleration / jerk values got corrupted during the FW change? I would try to reset the values to factory defaults.

closed 2019-01-29 by inquisitiveKangaroo