prusa3d/Prusa-Firmware #262: Feature Request: Let users choose which Temps to use during 1. layer calibration

MartinMajewski, 2017-11-01:
Hi all,

Because the first layer calibration is now part of the firmware, I think it is an excellent time to let the users chose which filament is used for this process. This possibility means that users should be able to choose the hot end and heated bed temperatures before the calibration starts.

The most apparent two ways of providing this selection would be:

1. Show the preheat menu, which is available anyway within the firmware.
2. Let the user sequentially adjust the temperatures, first for the heated bed, then for the hot end.

Currently, I use this feature less conveniently. I start the integrated first layer calibration, then I quickly navigate to the preheat menu and choose the PET settings.

Thanks for your efforts.

Bye,
Martin

MartinMajewski, 2017-11-01:
Edit: The reason this addition makes sense is, that not everyone uses PLA in a daily manner and want to buy/change-to PLA only for the sake of the calibration.

3d-gussner, 2017-11-02:
@MartinMajewski Most people use PLA and everyone has somewhere, and so the Wizzard using PLA calibration makes sense for beginners.

Everybody using ABS/PETG/HIPS/etc. should know what she/he is doing.

Here I think your idea would be great to provide different V2calibration_<Filament type>.gcode files.
As the Advance Linear Integration is enabled we even may need a different calibration procedure:
- Define which type of filament is used
- check https://github.com/MarlinFirmware/Marlin/issues/8079

Area5142, 2017-11-02:
I never use PLA, only prints with PET, ABS and Flex.
It would be nice to do materiel selecting from preheat menu right before the test print -  and have the k-factor test build into the firmware.

But as a more *advanced* user, I could live with a set of g-code files on on the SD card for different filament types and tests.

MartinMajewski, 2017-11-02:
Sorry 3d-gussner, but this is an unquantified generalization. Not even Prusa Research uses PLA and with all the filament compatibility advertised by Prusa for its printers it is not right to limit beginners to PLA but telling experienced users to use workarounds to prove their skills.

At least let all unsers see the preheat screen before the calibration. This screen is maintained by Prusa and should only contain temperatures/materials with which this test works. However, it sets the temperature at least somewhere near to the actually needed values of a particular material a user possibly uses.

Panayiotis-git, 2017-11-07:
I think it would be more complete, if after the user selects Calibration > First layer cal., the printer asks the user for the type of the loaded material. The choices screen should be a reuse of the Preheat screen.

MartinMajewski, 2017-11-07:
@Panayiotis-git At least we now know that this idea is a very obvious and relatively straight forward solution for this request! 🥇

joeynuggetz, 2017-11-14:
I think it would be cool if the single layer square that the new 1st layer calibration runs isn't near the edge of the bed. Code a nice Zhop over the existing test pattern lines and put that puppy near the center of the bed where most people usually print. This will never happen tho :-)

albegouws, 2019-05-29:
Bump - this will be great quality of life feature

draeath, 2019-09-06:
This is even more necessary now that we are required to perform a first-layer calibration when setting up new steel sheets in the HW Setup menu.

leptun, 2019-12-12:
@MartinMajewski The feature has been merged to the firmware. You can now close the PR.

MartinMajewski, 2019-12-13:
Thanks! Closed!

closed 2019-12-13 by MartinMajewski