prusa3d/Prusa-Firmware #267: Just upgraded to MM head and cannot get a good quality print.

jake9009, 2017-11-07:
Hi Guys,
I have upgraded my MK2S to the multi material mode and since then I cannot get it to successfully run through a print. I have tried just about everything. I upgraded to new firmware, I tried with ABS and PLA changed settings around tried printing old g.code that have been successful in the past on single print mode and am running out of ideas. It is almost as though the flow rate is wrong as the print inst sticking to the heat bed properly and as it builds up layers it stops printing or rips up the work with the pinda
[image]
[image]

 or the nozzle as the job is all skewed. I have attached a couple of pictures to help explain my issue. If anyone could please advise me on what to try I would really appreciate it as I am falling behind on some deadlines that I need the printer for.

macfly1202, 2017-11-07:
I encouter the sam eissue few weeks ago; Your first line has the big blob. Do you hear a big 'pushing' noise when the extruder start to print ?  Which firmware are you using ? 
I know we are on official Prusa thread but you could TESTthe 'community mod' (base on last RC2) I tested yesterday with this firmware for MMU (rambo13 :package only contain valid firmware for MMU, the other are the same) https://transfer.sh/ursZ2/prusa3d_fw_b0e2d1b.zip to know if not an issue with your Printer. Going from official to this release completely solve my printing issue (my K=200). Now I still need to test again the Prusa offical one to see if smoother like the community one.

macfly1202, 2017-11-07:
@Sebastianv650 : so the issue I encounter (big blob of filament at begining and huge push noise)   was link to official last RC1? Not yet test RC2 'official' but if you say it's the same features....Was is sure is something strange happen with firmware between firmware 3.012 single and MMU version and 3.1.0 RC1 that my mk2s mmu didn't like at all. This is now solved.

jake9009, 2017-11-07:
@macfly1202  That is correct i get the big blob and sometimes I hear that pushing noise. I tried lowering the Z axis a couple of times. I am not convinced that is the problem as it still hasn’t helped. Sometimes the job won’t print at all others it won’t stick to the bed. I have taken another two photos of the VM calibration print the first attempt only printed the purge line at the beginning the second attempt printed the job but it was very messy. We really need this printer to be operational as soon as possible as we have work pilling up
We are considering rebuilding it back to a single head.
We would still like to try to get it working but we are just running out of time. I know the product works as I have seen good reviews and amazing prints completed on YouTube.

Any advice would be greatly appreciated on where to go from here.
[image]
[image]

Sebastianv650, 2017-11-08:
Just some ideas due to the urgency:
When you load the filament by hand up to the nozzle, is this smooth or can you feel some areas where it is catching? 
Have you tried reflashing the fw and do a Factory Reset? 
Have you talked to the prusa support guys, maybe they have some quick ideas. 
Does this problem happen on every stepper, if you can try this? 
Have you disabled LIN_ADVANCE for test purposes?

macfly1202, 2017-11-08:
@jake9009 : you have a firmware / eeprom or slic3r issue. I had the same pb and have the printer like you. 
So my tips will be 

1/ test this parttern for PLA I made for MMU. It's print a single square on Extruder 3,2,1,3 (yes, not the 4th I use it for ABS only) 
https://www.dropbox.com/sh/g9ce9fhwo8e9dbu/AACto6mS0KKucgCtQea6c5fPa?dl=0
How is the result ? 
You must have something like this : 
[image]

2/ If not good, like sebastianv650 suggest, reset factory and reflash the FW with this one I tested and work good with mk2s+mmu like I write above
https://transfer.sh/12LKYQ/prusa3d_fw_4395775.zip (last RC2)
It's the version I use and tested ok with MK2S+MMU
(Always perform the calibration on Cold printer (It's a personal advice))

3/ Are you sure it is not an issue with slic3r and multi-material ? Had some issues in the transition between MK2S and the update to MMU with slic3r. Slic3r was keeping in memory standalone config of prusa and not taking in consideration MMU settings even if selected. I reinstall and delete all properties folder to solve this. No problem on my linux and my mac but on windows it's always complicated, unfonctionnal.

You could TEST with one filament loaded (aka not unloaded after a print OR manually loaded on printer with menu 'change extruder' and with 'Settings' 'Move axis' 'Extruder') : In slic3r, 'Print Settings' 'Multiple Extruders' Uncheck the box 'Enable' in the 'Wipe Tower' section. 
BE CAREFULL with this process ! it delete the wipe tower so your printer considers it's like a standalone MK2 not MMU. That means it will NOT load the MMU filament process. It's why you must have filament load first.  
 I use this trick during first print to calibrate Z, bed level correction etc... It avoids the boring steps (and not useful process) of loading each filament for finally calibrating on only one filament extruder... When you make 5-10-15 print for calibration, I assure you it's really quicker and avoid wasting filament ... 
Here a screenshot on windows slic3r version of this section you can disable for this test / calibration steps :
[image]

If still not ok, you could try with PrusaControl as slicerras a test :  https://prusacontrol.org/

Don't hesitate to ask, I love making test and setup again and again...

bubnikv, 2017-11-08:
> 3/ Are you sure it is not an issue with slic3r and multi-material ? Had
some issues in the transition between MK2S and the update to MMU with
slic3r. Slic3r was keeping in memory standalone config of prusa and not
taking in consideration MMU settings even if selected. I reinstall and
delete all properties folder to solve this. No problem on my linux and my
mac but on windows it's always complicated, unfonctionnal.

If you think there is an issue with Slic3r or the slic3r configuration,
would you please create a Slic3r issue and describe the issue with more
detail? Thank you.
https://github.com/prusa3d/slic3r/issues

macfly1202, 2017-11-08:
@bubnikv : the only way to test this again is to install a VM or find another Windows, install slic3r version (before the 1st slic3r MMU version THEN upgrade slic3r to the first version of slic3r mmu). Not really easily reproductible. Only concern windows (as always). Will try to test it again on WE but don't promise...

closed 2019-01-29 by inquisitiveKangaroo