prusa3d/Prusa-Firmware #269: 3.1.0 RC2: Pause/resume problem with Octoprint

Boby71, 2017-11-09:
Today, I had a very special problem:
I paused a print via Ocotprint to check something on a small test print.
The pause command worked normal: The print stopped, the print head moved up a (3-4cm?) - and went to the parking position.

But when I resumed the print, the head moved towards the printing object (correct) - and continued printing on its raised position, in the middle of nothing instead of lowering the head down to the printed object (not correct).

In Octoprint I do send the following commandos:
1) Pause:
M601 ; pause print

2) Resume: 
M602 ; resume print
G4 P5000 ; Dwell

Anyone else experienced a similar behaviour?

joeynuggetz, 2017-11-14:
I had this happen before but if I recall correctly it eventually lowers down to the right level? Not sure what version of firmware I tried this tho. I keep getting random disconnects from Octoprint and now even Astroprint so I haven't been printing much of anything lately but I'll try and test this.

Boby71, 2018-03-04:
Still, M601/M602 doesn't work in FW 3.1.0 as well on the MK2s.

Now it behaves the following way:
1) When receiving the M601 command (triggered by my filament runout sensor), the print stops and the head raises approx. 5mm
2) Then I did unload/load the filament procedure directly over the LCD panel
3) When resuming, the hotend temperature was set to ZERO(!)
4) Then the printer moved to X/Y 0 - and then tried to go down to Z0, which results into a crash in the printed part. Game over.

M601/M602 are completely useless (for more than a year now).

Prusa-Support, 2022-05-05:
Hello! If you are still having this issue, please contact us at info@prusa3d.com.

Marina Curzi
Prusa Research