prusa3d/Prusa-Firmware #276: 3.0.1 - Release - issues with display and Octoprint

MarcJenningsUK, 2017-11-12:
I just downloaded and flashed the 3.1.0 released tagged firmware (twice since I wanted to check for a bad download before raising an issue)
When printing from SD or Octoprint, the display does not show the target temperatures, but it remains at 0.  The display does show "Heating....." but actual temperature does not increase in the display either.

PavelSindler, 2017-11-12:
Which hex file version are you using? Can you send me gcode which you are using?
What do you mean by printing from SD? Did you start your print from menu or using Octoprint?

MarcJenningsUK, 2017-11-12:
I'm sorry, I'm out of the house now and will be for the rest of the day.
I'll send the details when I get back.

PavelSindler, 2017-11-14:
So can you please provide more information?

MarcJenningsUK, 2017-11-14:
Hi there,
I was just typing up a response.  It appears that the first file I tried on
both attempts at had become corrupt somehow and was essentially locking the
printer out in a temperature setting routine.  I have deleted the offending
file and retried with other files and am now successfully able to print on
the new firmware from Octoprint and the SD card.  Apologies for the
confusion.

PavelSindler, 2017-11-14:
And do you have some idea how it became corrupted? There is one issue which may be related to that: https://github.com/prusa3d/Prusa-Firmware/issues/234 and there was also similar issue opened in Octoprint repo.

MarcJenningsUK, 2017-11-14:
I think it may have been a manual error at some point.  I pinned it down to
the following lines of gcode

M104 205 ; set extruder temp
M140 50 ; set bed temp
M190 50 ; wait for bed temp
M109 205 ; wait for extruder temp

Note the missing "S" before the temperature values.

closed 2017-11-14 by MarcJenningsUK