prusa3d/Prusa-Firmware #280: Hosts using M110 are now broken with change between 3.1.0-RC1 and RC2. Issue remains in 3.1.0 Final.

JoshCraft3D, 2017-11-17:
Getting a constant error due to M110 command since firmware 3.1.0-RC2 and still with 3.1.0 Final.

M110 functioning without any errors on 3.1.0-RC1, and all previous firmware versions.
M110 errors appear with 3.1.0-RC2 and continue with 3.1.0.

Relevant Issue chain?
     Original Issue : #232
     232 Initial Fix : #235

In the #235 fix, the fix was then "deleted" due to an after comment from #232.

Console Log From Repetier-Server (Also Happens when connected through Repetier Host):
14:24:55.447: start
14:24:55.447: echo:echo: Last Updated: Nov 12 2017 11:07:38 | Author: (none, default config)
14:24:55.447: Compiled: Nov 12 2017
14:24:55.447: echo: Free Memory: 2158 PlannerBufferBytes: 1360
14:24:55.448: echo:Stored settings retrieved
14:24:55.448: M110
14:24:55.817: echo:SD card ok
14:24:57.051: ok
14:24:57.051: N2 M105
14:24:57.060: Error:Line Number is not Last Line Number+1, Last Line: 0
14:24:57.060: Resend: 1
14:24:57.060: Resend after 9196ms
14:24:57.071: ok
14:24:57.072: Resend: N1 M110
14:24:57.075: ok
14:24:57.075: Resend: N2 M105
14:24:57.084: Error:Line Number is not Last Line Number+1, Last Line: 0
14:24:57.085: Resend: 1
14:24:57.102: ok
14:24:57.102: N3 M115
14:24:57.113: Error:Line Number is not Last Line Number+1, Last Line: 0
14:24:57.113: Resend: 1
14:24:57.130: ok
14:24:57.130: N4 M220 S100
14:24:57.141: Error:Line Number is not Last Line Number+1, Last Line: 0
14:24:57.141: Resend: 1
14:24:57.158: ok
14:24:57.158: N5 M221 S100
14:24:57.170: Error:Line Number is not Last Line Number+1, Last Line: 0
14:24:57.170: Resend: 1
14:24:57.187: ok
14:24:57.188: N6 G92 E0
14:24:57.199: Error:Line Number is not Last Line Number+1, Last Line: 0
14:24:57.199: Resend: 1
14:24:57.216: ok
14:24:57.218: N7 M105
14:24:57.227: Error:Line Number is not Last Line Number+1, Last Line: 0
14:24:57.228: Resend: 1
14:24:57.245: ok
14:24:57.245: M117 IP:192.168.1.200
14:24:57.252: ok
14:24:58.060: N8 M105
14:24:58.071: Error:Line Number is not Last Line Number+1, Last Line: 0
14:24:58.071: Resend: 1
14:24:58.088: ok
14:24:59.067: N9 M105
14:24:59.074: Error:Line Number is not Last Line Number+1, Last Line: 0
14:24:59.078: Resend: 1
14:24:59.096: ok
14:25:00.076: N10 M105
14:25:00.086: Error:Line Number is not Last Line Number+1, Last Line: 0
14:25:00.086: Resend: 1
14:25:00.086: Resend after 3026ms
14:25:00.098: ok
14:25:00.098: Resend: N1 M110
14:25:00.103: ok
14:25:00.103: Resend: N2 M105
14:25:00.111: Error:Line Number is not Last Line Number+1, Last Line: 0
14:25:00.111: Resend: 1
14:25:00.128: ok
14:25:00.128: Resend: N3 M115
14:25:00.139: Error:Line Number is not Last Line Number+1, Last Line: 0
14:25:00.139: Resend: 1
14:25:00.157: ok

thess, 2017-11-17:
Not all hosts :grin: 

I do see the incompatibility. The single command ``M110`` correctly sets the last line number to 0. The command at: ``14:24:57.072: Resend: N1 M110`` should not have reset the last line number to 0 due to the lack of a following argument (and any reasonable specs). For compatibility with Marlin, etc. an M110 command with no argument is essentially ignored as long as the line number and checksum are correct.

Working on a fix to my original fix.

repetier, 2017-11-19:
Repetier always sends M110 with line number to set it. Line number comes before the M command but order of codes is not relevant in gcode, so that should not matter.

I'm not sure any more why we use 1 but there were problems with some firmwares if 0 was used.

mloidl, 2017-11-20:
Last weekend i could successfully test fix #283 with repetier server. Before this fix i also got this error.

repetier, 2017-11-20:
I don't think #283 is a full fix. It might work at startup but what is if you are at line 900 and send
M110 N1

or even M110 N43

would that also work? Since I have no prusa for testing I can't say for sure, but would be great if you could test this after some commands were send as well. We always send Nx as port of M110 so omitting the no N should not help. Might be that N comes after M if we force sending as is which could be the main problem.

repetier, 2017-11-20:
Looking into the server code I see we will always put N in front so it should detect the line number from testing line start so deleteing the line number to 0 should not be needed.

```
 	case 110:   // M110 - reset line pos		  	case 110:   // M110 - reset line pos
  		if (code_seen('N'))		  		if (code_seen('N'))
  			gcode_LastN = code_value_long();
-		else		
-			gcode_LastN = 0;
```
So question is more why code_seen is not working here.

thess, 2017-11-20:
@repetier - I believe all forms of M110 should now work. In fact as long as the checksum is correct when the M110 g-code is preceded by Nxx, even "Nxx M110 Nyy" will work setting the last line to 'yy'. If the M110 g-code has no preceding line number, it works as well. The handling of preceding Nxx to the M110 g-code is handled elsewhere in the parser.

JoshCraft3D, 2017-11-22:
Tried writing your fork for M110 fix, thess, but end up getting the following error on upload. It passes the compile verification, but fails on actual upload. This is the error:
"avrdude: verification error, first mismatch at byte 0x3c000
         0x80 != 0xa1
avrdude: verification error; content mismatch
avrdude: verification error; content mismatch."

It also was also failing at first because it was looking for a "hardware\marlin\avr\bootloaders\stk500boot_v2_mega2560" in the Marlin directory. Which wasn't in the Arduino addon's folder. I found a good hex for it, and it then was able to attempt to upload. Giving the above error.

*shrug*

thess, 2017-11-22:
@JoshCraft3D - I usually see this type of error when the wrong board config is set in the Arduino IDE. If you are interested in building your own firmware, you might want to checkout some revised build instructions I wrote a couple of weeks ago. Located here: [BUILDING.md](https://github.com/thess/Prusa-Firmware/blob/private-build2/BUILDING.md)

I also keep an up-to-date private build of the firmware with fixes and customized features. Pre-compiled .hex files are available in the same repo branch. Details here: [README.md](https://github.com/thess/Prusa-Firmware/blob/private-build2/README.md).

catrielmuller, 2017-12-01:
Sub

closed 2017-12-22 by PavelSindler

PavelSindler, 2017-12-22:
I have finaly tested it and merged it. Thank you @thess and thank you all for your patience. I have opened PR https://github.com/prusa3d/Prusa-Firmware/pull/343 for MK3 branch. It is the same code as in MK2, with just sligth modification as we there are two more bytes per command where command + comments length are stored (this is needed for power panic). I would be great if you take a look on this PR @thess . Thank you.

metacollin, 2018-03-31:
@thess That error has nothing to do with the board config. 

0x3c000 corresponds to the first address of the protected boot loader area of the ATMega2560's flash memory.  

The 'error' is just the intended behavior:  it isn't letting the boot loader get overwritten.  

Long story short,  whatever @JoshCraft3D did to the firmware, it made it too large to fit in the remaining flash memory.  The arduino IDE does not take into account the space consumed by the bootloader so it can say the sketch fits even if it doesn't actually fit.

JoshCraft3D, 2018-04-01:
All I had done was download the sketch of the fork thess had made for the fix and attempt to write it. So there must have been an issue with the fork, I didn't change anything in the fork once downloaded.

PR #283 (2017-12-22): Make M110 command compatible with Marlin, etc.
Don't reset line number to 0 if no argument to M110 G-code.

Fixes #280