prusa3d/Prusa-Firmware #281: Changing extruder mid print MMU

TheSappy, 2017-11-17:
A thing that could be great - possibilities to pause an MMU print and change extruder manually.
This could give an easy way to change material manually even in single material mode. Even a manual color change, if you don`t care about color bleed.

bubnikv, 2017-11-17:
The filament swapping sequence is a complex operation fine tuned for the
particular filament material. We would need the firmware know the sequence.
One way to do it is to have filament swapping "subroutines" stored in the
G-code, which would require both Slic3r and firmware changes as discussed
in https://github.com/prusa3d/Slic3r/issues/538

TheSappy, 2017-11-17:
I`m looking for something like
> Pause print
> Retract X length of filament from active motor
> Change extruder to X
> Load X length form active motor
> Resume print

A bit more complicated than stock mk2s Change filament feature.

bubnikv, 2017-11-17:
@TheSappy 
Your sequence is oversimplified and it will often fail. Our firmware pushes the molten filament out first, then it parks the active filament into cooling tubes, moves them a bit in the cooling tubes to not stick to them, and then first it loads the other filament.