prusa3d/Prusa-Firmware #282: SD Card Long Filenames broken again

IcicleTrepan, 2017-11-17:
SD Card cannot be read from OctoPi anymore again.

It looks like a previously applied patch to the cardreader.cpp file has been reverted 'because of compatibility' which appears to be causing this issue.  Please see https://github.com/prusa3d/Prusa-Firmware/commit/156aed0fdc09965335f8b7ac5e879278af857adb

I am not sure what compatibility problems it is causing but it would be nice to be able to use OctoPrint again..

thess, 2017-11-17:
The problem is not Octoprint. It lies with the SdFat library used by the firmware. Longfilenames (LFN) support is kind of bolted on as an afterthought and makes them available only as a side-status of the actual directory entry. You cannot use LFN's for opening or creating files. By reporting the LFN to Octoprint and then having it try to use it to in an "Print from SD" command, it fails. Hence, the need to revert to short filenames (8.3 DOS format) only on the serial interface.

BTW - The filename "xyz.gcode" is considered an LFN (extension is >3 chars)

ringerc, 2019-02-28:
See issue #1600

closed 2023-08-24 by github-actions[bot]