prusa3d/Prusa-Firmware #284: Really bad stringing.

jake9009, 2017-11-20:
Hi all,
Firstly I'd like to thank users @macfly1202 , @Sebastianv650 and @bubnikv for their help on my topic "Just upgraded to MM head and cannot get a good quality print." 
I was really busy with work so couldn't get back to you guys straight away however i did a factory reset uninstalled and re-installed all software and updated the firmware again and I started getting somewhere. 
It was a combination of that and i believe the pinda was not set correctly when the upgrade was originally done as I couldn't live adjust z to a satisfactory height. 

So I have completed a couple of successful prints since however am finding I am getting a lot of stringing. It wasn't to bad on the treefrog and it could easily be cleaned up, however i printed 3 small parts this morning and the stringing is really over the top. I was wondering if this is caused by my nozzle height or something else?

I have included a photo to show how over the top the stringing is.

Thanks again.
[image]

Sebastianv650, 2017-11-21:
Wow, that's the worst stringing I have ever seen! Please post your retraction settings, material type, nozzle temperature. Do you have LIN_ADVANCE in use?

Nozzle height is only important for the first layer, after that the effect of a wrong nozzle height goes to 0 within a few single layers. And you first layer seems to be quite good, so I wouldn't search there.

When I'm tuning retraction on a new printer from scratch I'm doing the following:
- Be sure the printer is set up correctly (steps/mm tuned, no play in bearings), test cubes print perfectly.
- Tune LIN_ADVANCE K factor (optional, if you want to use it)
- Enable retraction, start with default retract and prime speeds. Print something like your test pattern, while increasing retraction length and watch the differences. Be careful as there is also a too high retraction value possible. In this case, you can pull molten filament into the heat brake which will result in a jammed extruder. Stop increasing the retraction distance if you see no further improvement in stringing or when you are satisfied by the results.
- Next you can try to increase the retraction speed. Just increase the speed until the extruder can't catch up and is loosing steps, then lower by a not too small safety factor. This will also help with stringing. Note that you can't have a high prime speed, so if you go high on retraction speed you have to set a seperate prime speed some point. I recommend a slow prime speed anyway.

Reduction of nozzle temperature can help with stringing, but it will also limit the max. extrusion speed and result in higher needed K values due to higher nozzle pressure needed for extrusion. That's why I would stick with default temperatures as much as possible (205-210°C for PLA for example).

macfly1202, 2017-11-21:
same reaction : waoowww ! what is that ? on a prusa ? I never had such
stringing. One time, because too high temp (+20deg) on backup filament (Ice
Filament) on a 3 hour print had this at the end.
But this make think to something : not so good print last week. I print 3
vtech long train rail for my son and had similar string between the sharp
edge but not so much than your. here piece I print but have no photo.
https://www.thingiverse.com/thing:1452766
my temp was 200 degree.

wootner, 2017-12-18:
I get it also ....only since MMU upgrade ...reduced PL
[image]
A temp from 210 to 205 and 200 with little change
Also some Layer issue at a given height

bubnikv, 2017-12-20:
You may try the latest Slic3r https://github.com/prusa3d/Slic3r/releases
run the configuration wizard from the help menu to reinstall the bundled
printer profiles, which will update you the MM presets. This may improve
your linear advance and retraction settings, also there has been one
additional wipe added by Slic3r when moving away from the wipe tower.
Make sure you are on the latest firmware for the linear advance to work.

wootner, 2017-12-20:
I did call support who recommended changing the retraction from 4mm to 6mm. I did this and it did stop the stringing...I will try the latest Slic3r

closed 2022-05-05 by Prusa-Support