prusa3d/Prusa-Firmware #286: G80 disabbled with correct live adjust Z

Scrambler73, 2017-11-29:
I did some modifications on my Prusa MK2S where I managed to get a flat surface on the heatbed. This allows me to get a good first layer without using the mesh bed leveling (G80 G-code). This is a really big advantage for me because I use a glass plate for all of my prints and I don't need to remove and place the glass plate after the mesh bed leveling [(like Matt does on his tutorial)](https://mattshub.com/2017/04/11/printing-on-glass/). 

Ofcourse I need a reference point for the prinda probe, so I bought a aluminum sheet sticker and put a small piece on the corner of the glass. This is my point of zero for the Z-axle but now I have a problem that I can't repair.

My live adjust Z doesn't work correctly anymore. It works during the print where I can adjust the height but if I restart a second print it doesn't goes back to the live adjusted Z height. It goes back to it's original zero and I need to adjust the height again in the live adjust Z. I also see that the value has been saved from the previous print forcing me into doubling the value to lower the nozzle. 
I would like to restore this but I think I need to adjust the firmware of the prusa, and I don't have any knowledge on that subject.

3d-gussner, 2017-11-29:
@Scrambler73 you can always do a factory reset
http://help.prusa3d.com/38908-firmware-software/276545-factory-reset

Scrambler73, 2017-12-05:
How would a factory reset resolve my problems? And I don't wanna lose my statistics on the Prusa, I guess when I would reset this it will start again from zero.

Scrambler73, 2017-12-08:
Well, I did the factory reset. The statistics on my Prusa did reset what I was afraid of. I would suggest that this would be changed, there is no valid reason to reset the statistics and people will only abuse this. People can reset the statistics on there Prusa and sell it as brand new. If you had this option on a car then fraud will be beyond doubt. 
I can honestly say, when the statistics can be reset the whole statistics is worthless because it says nothing when a reset can be applied.

But back op topic!
After the reset I didn't do the calibration. Now after I adjust the Live Z the values changes back to zero, with that now happening I'm satisfied.

GurliGebis, 2017-12-09:
You need to do the V2 calibration for it to save the Live Z values :-)

closed 2022-05-05 by Prusa-Support