prusa3d/Prusa-Firmware #293: MMU wait for user issue.

totalitarian, 2017-12-06:
Why do multi colour prints pause occasionally with the prompt on the lcd that says "wait for user...."

Tried to do a multi colour print last night and the print head stayed parked over the priming tower for hours on end. I tried to resume the print in the morning but the nozzle had melted the prime tower after being parked.

bubnikv, 2017-12-06:
This wait is dictated by the Slic3r for you to inspect that all the filaments printing were primed correctly.
In case the priming towers collide with the object, Slic3r will wait for you to remove the priming towers and confirm.

We should improve Slic3r & firmware to provide a descriptive message on the printer display though.

totalitarian, 2017-12-06:
Thanks but that means I can't leave my prints unattended and with the length of time it takes print multi material, this is a deal breaker for me :(

bubnikv, 2017-12-06:
The printer shall not stop during the print. Old firmware used to display "Wait" on a G4 G-code, which we use to flush the firmware planner queue at some point of time over the wipe tower, but that call should not stop the print.

Are you printing from a SD card or from an OctoPrint or Pronterface or any other application over the USB?

totalitarian, 2017-12-06:
I'm printing from SD card, It's definitely pausing and waiting for user input when sat over the prime tower several times during a print

cosmith71, 2017-12-23:
I had this happen today.  Approximately 9 hours into an SD print and the print head had stopped over the priming tower and was displaying "wait for user".  I'm not sure how long it had been that way (I hadn't checked on it in about 2 hours).  I was able to click the LCD knob and it started up again.

cosmith71, 2017-12-23:
[image]
[image]

cosmith71, 2017-12-24:
[image]
[image]

ilusion316, 2017-12-25:
Same experiences here, please check https://shop.prusa3d.com/forum/original-prusa-i3-mk2-multi-material-f26/-wait-for-user--t6774.html for details. I believe there is an association with octoprint being connected and its communication with the printer while wiping.

totalitarian, 2017-12-27:
I didn't have octoprint connected

Prusa-Support, 2022-05-05:
This issue will be closed due to it being obsolete. In case the issue is still not solved, please contact info@prusa3d.com.

Marina Curzi
Prusa Research

closed 2022-05-05 by Prusa-Support