prusa3d/Prusa-Firmware #301: Live adjust Z needs to be done for every print

aherrone, 2017-12-10:
After updating to PT100 temperature sensor a firmware reflash was needed and since that every new print needs Z offset adjusted. For example if the first print is fine with -0.500 adjustment, then the second time a print is started the value needs to be adjusted from -0.500 to -1.000. Then again third print needs a value of -1.500 and so on.

Because we changed the nozzle to a 0.50 mm one we are not using the original V2Calibration.gcode supplied with the printer. Instead we just tune the Z during a large objects first layer. The printer is MK2S.

PavelSindler, 2017-12-22:
Live adjust Z values are not stored to eeprom unless G87 gcode is executed. This gcode is part of V2Calibration.gcode file. If you haven't print this file, just send this gcode over serial or add it manualy to some gcode file. Let me know if this solved your issue.

aherrone, 2017-12-22:
Well thanks for the reply. I didn't know that the setting needed to be saved to eeprom every time. I haven't tried the G87 yet, but if this is the way it works then it really should be improved. My problem was that even when the printer wasn't power cycled, the setting had to be changed (add the correct offset value to the wrong value showed on the screen) for every print. Without looking into the code I would say that there are two different values for this setting - one that is shown on the screen and one that is actually used. I think this is a major problem. 

The reason the V2Calibration.gcode was not run was that the nozzle was changed to 0,5 mm. Should I do the calibration with the gcode for 0,4 mm nozzle or is there a way to generate a calibration gcode for any nozzle size?

clearbucketLabs, 2018-01-30:
It's a bug when the offset on the LCD isn't the one it's using....  if the in-memory LIVE Z is 0.5mm,  the next print you shouldn't have to change it to -1mm to get the SAME hight as the previous print... then -1.5mm on the print AFTER that one.... that's a bug....

Crunch69, 2018-03-14:
still an issue

SteveQUT, 2018-11-13:
Yep still an issue.

Boby71, 2019-01-08:
> Yes. Still an issue. Spend a good 2 hours to find out the cause. Thought it was my probe. Indeed I recalibrated and did not run the calibration gcode

Lucky guy that you found it out after two hours...it took me three days since I never used the built-in calibration pattern (because I always use a 50x50mm square to calibrate my first layer (adjusting the live-Z during the print; after one turn everything is fine) - which I find much better to identify the quality of the first layer instead some tiny lines and a 10mm sqare)

gudnimg, 2023-07-03:
I think this issue was fixed a long time ago. Closing this Github issue as it seems to be stale at this point.

closed 2023-07-03 by gudnimg