prusa3d/Prusa-Firmware #318: File sorting broken when using Toshiba FlashAir SD WiFi

Area5142, 2017-12-13:
Sorting on SD card is not updated when adding/removing files from Toshiba FlashAir SD card via WiFi.
New file are sometimes misplaced and deleted files sometimes appear in 8.3 file format with first letter replaced by '~' (old DOS way of telling that the file was deleted).

Firmware 3.1.0, RAMBo13a, MK2S SM.

johnlaur, 2017-12-23:
The sort table is precomputed upon the card insertion; the firmware isnt written with the expectation that the filesystem on the card will change unless acted upon by the firmware. I suppose if there were a option to turn sorting off completely and nothing about the filesystem was 'cached' then things would behave better. Sorting is disabled for filesystems with >100 files so a temporary workaround might be to put a lot of files on your SD card in a subdirectory to force sorting to turn off.

Area5142, 2017-12-23:
The firmware supports FlashAir cards. There must be a way that the it can detect that the files/directories have changed via a timestamp/checksum or similar and then recreate/rebuild the 'cache' when needed.
I don't want the sorting to be disabled, just updated.
My workaround is to put very few files in a sub directory, then it is easy to find the right one.

cjparsons, 2018-02-15:
I agree - if there is an option to use FlashAir cards then surely they can re-sort whenever a file is added/changed?

SD cards are a bit fragile and the idea of having to haul it out and put it back in every time I want a new print I believe lets the printer down...

ochm, 2019-03-03:
Every time user open directory (folder), the files should be sorted first.

closed 2023-08-24 by github-actions[bot]