prusa3d/Prusa-Firmware #330: MK3 firmware not communicating via UART

KissmanCZ, 2017-12-16:
I have connected Rpi zero with EINSY via UART. Octoprint can connect but there is timeout from printer.

`Connecting to: /dev/ttyAMA0Changing monitoring state from 'Offline' to 'Opening serial port'Connected to: Serial<id=0xacba7790, open=True>(port='/dev/ttyAMA0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorChanging monitoring state from 'Opening serial port' to 'Connecting'Send: N0 M110 N0*125There was a timeout while trying to connect to the printerChanging monitoring state from 'Connecting' to 'Offline'Connection closed, closing down monitor`

Same for Repetier Server.

Allram, 2017-12-16:
Tried with baudrate 125200?

KissmanCZ, 2017-12-16:
Yup... same :(

`Connecting to: /dev/ttyAMA0Changing monitoring state from 'Offline' to 'Opening serial port'Connected to: Serial<id=0xad0bc4d0, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorChanging monitoring state from 'Opening serial port' to 'Connecting'Send: N0 M110 N0*125There was a timeout while trying to connect to the printerChanging monitoring state from 'Connecting' to 'Offline'Connection closed, closing down monitor`

KissmanCZ, 2017-12-16:
Same

`Connecting to: /dev/ttyAMA0Changing monitoring state from 'Offline' to 'Opening serial port'Connected to: Serial<id=0xad06e450, open=True>(port='/dev/ttyAMA0', baudrate=125200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorChanging monitoring state from 'Opening serial port' to 'Connecting'Send: N0 M110 N0*125There was a timeout while trying to connect to the printerChanging monitoring state from 'Connecting' to 'Offline'Connection closed, closing down monitor`

KissmanCZ, 2017-12-16:
Just tried disable internal bluetooth from ttyAMA0 and disabled console on UART... but same :(

`Connecting to: /dev/ttyAMA0Changing monitoring state from 'Offline' to 'Opening serial port'Connected to: Serial<id=0xac95c730, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorStarting baud rate detectionChanging monitoring state from 'Opening serial port' to 'Detecting baudrate'Trying baudrate: 250000Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 125200Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 115200Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 230400Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 57600Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 38400Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 19200Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 9600Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Changing monitoring state from 'Detecting baudrate' to 'Offline'Changing monitoring state from 'Offline' to 'Error: No more baudrates to test, and no suitable baudrate found.'Connection closed, closing down monitor`

Allram, 2017-12-16:
And you are sure you are using the correct pins? I think they got changed from what they originally was.

KissmanCZ, 2017-12-16:
Yes, I am sure... I cannot use wrong pins, because printed frame has predefined holes for them.

[image]

KissmanCZ, 2017-12-16:
https://github.com/foosel/OctoPrint/issues/2293 :( Prusa team, pls release this feature ASAP. Thanks!

PavelSindler, 2017-12-18:
Hello @KissmanCZ , which fw version do you have?
It should work in RC2 version which can be downloaded from our website https://www.prusa3d.com/drivers/

PavelSindler, 2017-12-18:
In RC2 firmware version, you must enable using second USART in settings menu.

PavelSindler, 2017-12-18:
We are currently also working on some basic guideline how to setup it. It will be posted here soon.

tibor-vavra, 2017-12-19:
## OctoPi on Raspberry Pi Zero W with Original Prusa MK3 HowTo
------

- Download OctoPi img
  https://octopi.octoprint.org/latest

- Unzip it

- Download Etcher tool
  https://etcher.io/

- Write it on micro sd card for RpiZeroW by Etcher

- Edit config file on written sd card filesystem **octopi-network.txt, uncomment and edit section what you want to use**

[image]

  This example will add **OctoPi** to wi-fi network **"testik"** with password **"12345678"**

  ​

- Put it in your printer, turn on and set in menu **Settings -> 2nd USART [on]**

- Find **ip adress** of this **OctoPi** in network

- Connect on it by ssh (default user is **'pi'** and password is **'raspberry'**)

- Run: **sudo raspi-config**

- In menu **Interfacing Options -> Serial** -> on question **'Would you like a login shell to be accessible over serial?'** answer **No** and on question **'Would you like the serial port hardware to be enabled?'** answer **Yes**.

[image]

[image]

[image]

[image]

[image]

  Press **Finish** button and **reboot it**

  ​
- Go to **ip address** of **OctoPi** in your web browser to **settings option(wrench icon on right top)** and add new serial port **/dev/ttyS0** and baud rate to **115200**. 

[image]

[image]

- After **reset** you will be able to make connection to the printer 
[image]

Pretender1988smg, 2018-01-14:
HI i tried exacly the same and i still get a timout.... Did some one got it to work?

AustinSaintAubin, 2018-10-06:
https://help.prusa3d.com/l/en/article/Loz15FAgEk-octo-print-raspberry-pi-zero-w

adamit79, 2019-06-05:
thank you, it worked

closed 2023-08-24 by github-actions[bot]