prusa3d/Prusa-Firmware #333: Bed leveling adjustment - corners instead of axes

mareklinka, 2017-12-19:
While calibrating our new MK2S I found an issue when while the center of the bed was calibrated pretty well (with a consistent layer height), the outside areas were skewed from top-right to bottom-left.

So I came up with a calibration print that prints 2x2cm single-layer squares very close to the corners of the bed. I reasoned that as long as the PEI board is reasonably flat, calibrating the corners to give consistent first layer should provide the same everywhere else. After playing around with this for a few hours I achieved pretty good results with this method.

However, this way of calibration is not easy to do because the bed leveling adjustment works on left/right/front/back axes. Therefore moving front-left corner up while keeping the front-right corner as-is (because it so happens that it's calibrated correctly) is difficult.

I would therefore propose the bed leveling adjustment to work over corners instead of sides. Is it something that would be possible/desired?

Panayiotis-git, 2017-12-19:
Hi,
have a look at the:
https://github.com/prusa3d/Prusa-Firmware/pull/220

With this PR I finally managed to print to the entire bed without problems. It is an amazing solution!

mareklinka, 2017-12-19:
Awesome! I really hope this gets merged soon. It would make the calibration much faster.

closed 2022-05-05 by Prusa-Support