prusa3d/Prusa-Firmware #360: Pinda Heating

martinowren, 2017-12-29:
Maybe a good choice to make the pinda heating 60 seconds instead of 120? A bit too long to wait I feel. 

Maybe it would be a better idea to do the mesh bed leveling before heating the bed and extruder?

Panayiotis-git, 2017-12-29:
It is indeed too long to wait.

I wish this experimental feature to work reliably for you.

But, for me it is a dangerous feature. I started to use it and for some time it worked fine. Then, after I was relaxed and confident about it, one time the hot end became an excavator...

Prusa-Support, 2022-05-05:
This issue will be closed due to it being obsolete. In case the issue is still not solved, please contact info@prusa3d.com.

Marina Curzi
Prusa Research

closed 2022-05-05 by Prusa-Support