prusa3d/Prusa-Firmware #362: Repetier Server

strongbow1, 2017-12-29:
Repetier Server cannot communicate with the printer. It connects but will not send anything to the printer. The older version of the firmware did work. could you please address this issue I have to revert to astroprint or octoprint for the time being

KissmanCZ, 2017-12-31:
Which version of firmware, which printer, which connection? I am using latest on my MK3 with GPIO & USB too and all working fine with latest Repetier Server.

strongbow1, 2017-12-31:
Prusa i3 Mk2s using the firmware 3.1.0 the latest one you can download from the Prusa support web page and I am using repetier server vers 0.86.2

KissmanCZ, 2017-12-31:
So u must read better ;) there is a notice on Prusa driver page:

> There’s a known bug using Repetier server, please wait for the next release if you’re using it

So, wait for the next release or downgrade ;) for a while...

cboileau, 2018-08-18:
Is there ever going to be another firmware update with a fix for this? It's been broken since November with no update

3d-gussner, 2018-08-19:
@cboileau Which firmware and printer do you use?
@strongbow1 The MK2 branch merged a serial communication fix on 17. Nov 2017 but the latest release hex files you can get from Prusa is from 12. Nov 2017 so you it is missing that fix.

Try to compile the MK2 branch for yourself or checkout my https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.1.0-001-HFM for the MK2/s/MMU