prusa3d/Prusa-Firmware #370: MK3 Filament sensor N/A on b143

zafuquazar, 2018-01-04:
On 3.1.1 RC2 and RC3 firmware, my mk3 filament sendor worked fine. On RC3, the auto load function worked well.
When I upgraded to RC4 (b143), my filament sensor now reports as [N/A]. I have not touched any wiring or connectors. I suspect something in the b143 code is causing my sensor to no longer be seen by the controller. Thanks!

PavelSindler, 2018-01-05:
Thank you for reporting this. It is bug. For now, turn it on in settings menu again and it should work.

zafuquazar, 2018-01-05:
Thank you. Actually, I cannot turn it on in settings. When I try to turn it on, I get the error “ERROR: Filament sensor is not responding, please check connection”. As I said, I did not change any wiring or connections other than flashing to b143.

zafuquazar, 2018-02-09:
I confirm that the filament sensor continues to report as N/A in RC5 and 3.1.1 final

zafuquazar, 2018-02-09:
I will try a factory reset (as suggested by Prusa chat) and see if that fixes it

easycheese, 2018-06-24:
@zafuquazar did that fix it?  I'm having the same problem

closed 2023-08-24 by github-actions[bot]