prusa3d/Prusa-Firmware #371: Wording of "filament clear" is awkward in b143

zafuquazar, 2018-01-04:
After upgrading to b143, the on-screen text (English language) after loading filament reads "Filament extruding & with correct color?" Yes/No.
This grammar is awkward in english. I suggest to change the wording to something like "Is filament color correct?"
I understand the change from the old "color clear", since "clear" can be misleading. But it is the addition of the "Filament Extruding &" that is awkward. If the filament has not started coming out of the nozzle yet, then "Is filament color correct?" is false and the user can choose No.
Thanks,

Panayiotis-git, 2018-01-04:
If I use multiple materials of the same color?

zafuquazar, 2018-01-05:
If you are using multiple filament of the same color, then you don’t need to purge much to be happy and you can say yes to “Is color correct?” on the first try :)

zafuquazar, 2018-02-09:
Wording in 3.1.1 final is great. Thank you!

closed 2018-02-09 by zafuquazar