prusa3d/Prusa-Firmware #372: 3.1.0 is non functional on my i3 mk2s

jmtrappier, 2018-01-04:
After upgrading to 3.1.0 and calibrating the printer, every attempt to print anything makes the nozzle to push the bed. First attempt even damaged it.

Please, remove 3.1.0 since it is not functional, I reverted to 3.09 and everything went fine, I then retried to upgrade to 3.1.0 and it failed again.

Not sure why I am alone with this but this is a serious issue because printer is damaged due to this firmware.

GurliGebis, 2018-01-05:
It is working fine on my printer.
Could you try and do a factory reset, and then a full calibration, to see if it still fails? (Remember to move the print head all the way to the top).

Panayiotis-git, 2018-01-05:
It is working fine for me too.
There are too many official releases (final and candidate) between 3.0.9 and 3.1.0. In fact 3.0.9 is not even available for download.
Could you download and flash a later one and try to find in what release your problem appears?

jmtrappier, 2018-01-08:
You can download 3.0.9 (I did so you can).

I tried what GurliGebis said and yes, this works, I had to factory reset the printer and now 3.1.0 is working "fine".

Waiting for multi material upgrade to test it but now I know that a factory reset is required when you upgrade firmware, strange.

GurliGebis, 2018-01-08:
Normally a factory reset shouldn't be needed, but sometime the values stored in the EEPROM can get mangled, which a factory reset fixes. I haven't had the need to reset mine yet, so it is the exception, not the rule that it must be reset :)
Often it is easier to do a factory reset followed by a complete recalibration, since it help ensure the problem at hand is not caused by bad values.

jmtrappier, 2018-01-08:
It is still very strange PINDA didn't prevent the nozzle to damage the bed, it was really pushing on it and tried to move meanwhile, scratching severly the bed. Hopefully, it is a the edge so it shouldn't be a big issue but still a bug that did this, not a bad move from me...

closed 2022-05-05 by Prusa-Support