prusa3d/Prusa-Firmware #373: PINDA temperature calibration gone bad

stahlfabrik, 2018-01-05:
Hi,

my PINDA probe sensor on my MK3 got heavily temperature dependent since a few days - it might have coincided with the latest firmware release. I also did a factory reset and also tried a temperature calibration. To find a solution I did again a factory reset and reflashed the firmware.

But it does not help: the warmer the PINDA gets the more squished the layers get. And when the PINDA is cold, the layer height is way to high.

Is there a way to show the internal temperature calibration table that my EINSY uses? How can I reset this table so that it has default values? What could I do see if it is hardware issue (the PINDA generally works and shows sane temperatures under support/temperatures)

On the picture: left the PINDA was warm after a print (layer slightly to squished). Right it was room-cold (light shines through space between extrusion lines)
[image]

Best regards

bubnikv, 2018-01-05:
https://github.com/ultimachine/Einsy-Rambo/blob/1.0a/board/Project%20Outputs/Schematic%20Prints_Einsy%20Rambo_1.0a.PDF
sheet 6 shows the connection to the PINDA probe.

A good test is to measure the resistance between the two out-most pins of the cable. At room temperature the resistance should be around 120 kOhms. At 25 degrees, the resistance falls down to 100 kOhms.

stahlfabrik, 2018-01-05:
Ok thank you! I will measure later when I am home. In the meantime: a factory reset will reset the temperature calibration table and bring the rest of the firmware to use the default table again?

stahlfabrik, 2018-01-05:
Hi @bubnikv 

So I measured between the two out-most pins (1 and 4 in the schematics - colors brown and white on the cable). The resistance was bigger than my multimeter can show. So I tried other pins. Between the pins 1 and 3 there is a variable resistance - colors white and blue). I measure 124kOhms at 19.3 degrees. I put two finger tips around  the PINDA and the resistance was going down to 89kOhms. 

So is my PINDA maybe not wired correctly? Are you sure about the two out-most pins?

And could it be a firmware bug, because the EINSY does read temperatures from the PINDA. Could I dump the temperature to Z adjust table somehow to see what is being used? Could I have deleted it through a temperature calibration? I think I even canceled an temperature calibration once - but I do not know if that was before I had the problem (I guess I started looking into temperature calibration because of my problem.) 
Would performing a factory reset or reflashing the firmware recreate this table?

Thanks for your help!

stahlfabrik, 2018-01-05:
I did another factory reset. Then I did now a M502 then M500 to reset EEPROM. 

Still the PINDA probe is totally sensitive to temperature. I don't know what to do. How could I be sure that the EINSY has sane / default values in its EEPROM for the temp correction of z adjust?

stahlfabrik, 2018-01-05:
void Config_ResetDefault() seems to NOT reset temperature calibration default values! (But I must confess that I look at Marlin Code for the first time. Is it possible that I messed up my calibration table and now cannot get it back to true factory defaults?

bubnikv, 2018-01-06:
Sorry, I made a mistake. The thermistor is connected between the blue and white wire. If it does not read around 120 kOhms at 20 degrees centigrade, then you need a replacement probe to be ordered over our support.

[image]

stahlfabrik, 2018-01-06:
@bubnikv

My probe reads 124k at 19.3 degrees and in the menu it shows temperatures that are comparable to the other thermistors. 

How could I reset the temperature calibration table to factory defaults? My guess is that here lies my problem. 

How can I execute the D8 ? and D8 ! Debug codes?

Please consider reading my last nights comments above - i fear my einsy might need to be replaced if i am right that my calibration table is bad and cannot be reset.

jettoblack, 2018-01-28:
@stahlfabrik Sounds like you might be having the same problem as some of us. Please see this thread:
https://github.com/prusa3d/Prusa-Firmware/issues/421
and this potential fix by TheZeroBeast: https://github.com/prusa3d/Prusa-Firmware/pull/439

TZB built a firmware with a new PINDA temp calibration routine and it has greatly reduced my calibration issues. It also lets you print the temp calibration table via G75. There was a huge difference in my temp calibration table before & after his fix and it seems to have done the trick.

stahlfabrik, 2018-01-29:
@jettoblack
Yes I am aware of th potential improvement by the other guy’s pull request I have yet to try it. 

I just wonder if it can be it. Because after factory reset the temperature calibration table is restarted with the same default values for every body. Temperature calibration is an optional thing to do , right?

But awesome that the other user has put a lot of work into improving this. 

I still have a replacement pinda here that I have yet to install. 

By the way: we both are in the same Facebook chat that Eric has started. 

According to his experience there is nearly zero chance that swapping the pinda will help anything. 

No good look out for me. Will try the new pinda soon but first I will try the proposed firmware

stahlfabrik, 2018-02-03:
Today I installed a replacement pinda that Prusa research support has send me. 

The problem continues to be there. A good first layer is not repeatable: each time life z has to be tuned in. And when the printer cools down a little between prints the next print starts to high. When you start a new print quickly the print is lower than before. 

Is this a software bug? Or is my einsy or heatbed or powersupply the problem?
 
Pinging @bubnikv

I installed latest firmware from yesterday and did a factory reset.

bubnikv, 2018-02-03:
You may want to run the PINDA probe temp calibration. Please install the
latest firmware and run the temp calibration with the steel plate removed.

stahlfabrik, 2018-02-04:
@bubnikv I did the temperature calibration but it did not help. The ambient temperature during temperature calibration was a stable 21C.

Here is a photo
[image]
 of the first layer of Prusa logo. The upper print was the result from 3 or 4 tries (without letting the printer cool down) to get a nearly good first layer. On the lower print it is the same live z value. The difference is that I let the printer cool down to 25C before I started the print.

Would you be interested to get hold of my MK3? I contacted your colleague Jennes about that. And I would be happy to receive working components in a fresh kit.

It seems that this bug affects only a handful of users. I heard from one where swapping the PINDA and also swapping the EINSY did not help. 

I am in despair and do not know what could be the problem here.

Firmware is 3.1.1-198
Factor reset after upgrade
Self test passed
XYZ calibration all fine (let 9.88mm right 9.69mm; Skew 0.00 degree)
I did a PID calibration - did not improve things
I did a temperature calibration - did not improve things

3d-gussner, 2018-02-05:
@stahlfabrik What values do you get from gcode D8 and M75?

stahlfabrik, 2018-02-05:
@3d-gussner I will check that out later when I am home.

@bubnikv 
Meanwhile I forgot to mention my G80, G81 numbers - Are these supposed to be somehow stable when the ambient temperature changes? Because they are very different here when I compare the 16C ambient temp to 20C ambient temp values. If that is NOT normal than there would be "naked numbers" that show the huge influence of temperature that I experience.

These numbers were read before a temp calibration - aka with default temp calibration values:

Cold, 16C hotend and bed:

Measured points:
  0.52250  0.42306  0.34556  0.29000  0.25639  0.24472  0.25500
  0.52667  0.43127  0.35059  0.28463  0.23340  0.19688  0.17509
  0.50389  0.40840  0.32355  0.24935  0.18580  0.13290  0.09065
  0.45417  0.35444  0.26444  0.18417  0.11361  0.05278  0.00167
  0.37750  0.26941  0.17327  0.08907  0.01682  -0.04349  -0.09185
  0.27389  0.15330  0.05003  -0.03593  -0.10457  -0.15590  -0.18991
  0.14333  0.00611  -0.10528  -0.19083  -0.25056  -0.28444  -0.29250

20C hotend and bed
  Measured points:
  0.40750  0.35380  0.30296  0.25500  0.20991  0.16769  0.12833
  0.43278  0.37001  0.30828  0.24759  0.18794  0.12933  0.07176
  0.43028  0.35794  0.28612  0.21481  0.14402  0.07374  0.00398
  0.40000  0.31759  0.23648  0.15667  0.07815  0.00093  -0.07500
  0.34194  0.24896  0.15936  0.07315  -0.00968  -0.08913  -0.16519
  0.25611  0.15205  0.05476  -0.03574  -0.11946  -0.19641  -0.26657
  0.14250  0.02685  -0.07731  -0.17000  -0.25120  -0.32093  -0.37917

After temp calibration 21C hotend and heatbed
Measured points:
  0.46917  0.40639  0.35111  0.30333  0.26306  0.23028  0.20500
  0.47417  0.41571  0.35802  0.30111  0.24497  0.18960  0.13500
  0.46111  0.39997  0.33645  0.27056  0.20228  0.13164  0.05861
  0.43000  0.35917  0.28639  0.21167  0.13500  0.05639  -0.02417
  0.38083  0.29330  0.20784  0.12444  0.04312  -0.03614  -0.11333
  0.31361  0.20238  0.10080  0.00889  -0.07336  -0.14596  -0.20889
  0.22833  0.08639  -0.03472  -0.13500  -0.21444  -0.27306  -0.31083

When repeating G80, G81 quickly five times for the same ambient temperature there is only very little variation!

stahlfabrik, 2018-03-03:
@bubnikv
I spent many days and hours investigating and getting familiar with parts of MK3’s firmware. 

I found a small bug which seems to cause the issue. I issued a pull request which fixes the issue. 

Please have a look at my pullrequest #514

stahlfabrik, 2018-04-27:
Happy to say that 3.2.0-RC2 SHOULD fix the issue (by supporting negative ustep offset values)

Thank you Prusa Research for adopting my work on that issue and improving it further!

closed 2018-04-27 by stahlfabrik