prusa3d/Prusa-Firmware #375: MK3 Print from sd freeze

alessandroame, 2018-01-06:
With 3.1.1 b143 RC4 when I print from sd often the printer freeze (with bed and hotend kept warm) until I press one time the knob.

PavelSindler, 2018-01-08:
Hello @alessandroame , do you have some gcode file where this issue occured? Can you send it here?

alessandroame, 2018-01-08:
Yes I've gcode but is a commercial model so I must send it privately. How can I do that?

DiddyWolf, 2018-01-16:
I've also seen this issue on 3.1.1 b143 with the gcode attached.
[Speaker_Mount_v2.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/1633684/Speaker_Mount_v2.gcode.txt)

Prusa-Support, 2022-05-05:
This issue will be closed due to it being obsolete. In case the issue is still not solved, please contact info@prusa3d.com.

Marina Curzi
Prusa Research

closed 2022-05-05 by Prusa-Support