prusa3d/Prusa-Firmware #376: MK3 Tranparent TPU and filament sensor

alessandroame, 2018-01-06:
With 3.1.1 b143 RC4 when I print transparent tpu (I've tried all sainsmart transparent colors) often the print stop, the bed go back the extruder go right and the printer ask to unload and then reload the filament.
When I disable the sensor or when I use pla with solid color the problem does not persist.

cnlson, 2018-01-17:
i've had this happen a couple of times with solid colors. but it appears there might have been some over-retraction going on which caused it to need to be restarted.  Conversely I have printed Sainsmart transparent Pet-g with no issues at all

funkycode, 2018-03-04:
I printed with white e3d edge filament (PETG) with no problem, while with black esun PETG I get constant requests to load and unload filament. 
Little bit hard to tell if it's coz of low temp, jams, retraction or colour.

jolcese, 2018-03-06:
I have the same problem with transparent Sainsmart TPU and firmware 3.1.2

renarsvilnis, 2018-05-26:
Currently printing a model with transparent Flexfill 98A, every 10-20min I get the issue where I need to unload and load the filament. It's so annoying, as I'm in 3h of printing.

First time using flexfill - is it that it's a quirk with the filament itself or the filament sensor struggling with transparent filaments?

Using `Prusa mk3` with `v3.2.1` firmware.

westranm, 2018-06-07:
I'm having this same issue with eSun transparent florescent orange PETG on v3.2.0 firmware.  However, I have used clear, florescent green, and red transparent PETG filaments from YOYI without any issue.

closed 2023-08-24 by github-actions[bot]