prusa3d/Prusa-Firmware #378: New MK3 Firmware 143e causes X and Y belts to slip.

JagerSprinkles, 2018-01-06:
I never had an issue with the X and Y belts slipping before I updated. The Y belt always slips right after the purge line when the head goes to the first point of the print. Later the X and Y belts will slip when doing a travel move. The slip happens when the head decelerates at the end of the move like the stepper is decelerating too fast.

I have reverted to firmware r137 and the problem is completely gone.

Some more info:
Printing from Octoprint 1.3.6 via USB
Same G-Code files slip on r143 and not on r137

I would have posted this to the Prusa forums but I am unable to log in there.

craftit, 2018-01-15:
I've seen this issue on a brand new factory assembled printer, both printing from USB as well as from SD card though it took longer to happen.  Using lower poly versions of the parts, (which generate smaller gcode files) seems to reduce the chances of it happening.

JagerSprinkles, 2018-01-21:
This is still a problem with version 150

closed 2019-01-05 by JagerSprinkles