prusa3d/Prusa-Firmware #381: How to use Serial2?

probonopd, 2018-01-08:
For printing wirelessly, I have connected an ESP8266-based device to the MK2S and written https://github.com/probonopd/WirelessPrinting.

In order for this to work, I need to use Serial2 instead of Serial0 for communicating with the printer and for sending GCODE (connector P3 on the RAMBo instead of the USB conncetor).

Earlier this year, I used to do:

```
sed -i -e 's|SERIAL_PORT 0|SERIAL_PORT 2|g' Firmware/Configuration.h
```

It used to work, e.g., in b3eeac1.

Now, e.g., in f91ace6 it seems to no longer work and gives an error:

```
In file included from /home/travis/build/probonopd/Prusa-Firmware/sketch/Marlin.h:18:0,
                 from /home/travis/build/probonopd/Prusa-Firmware/sketch/MarlinSerial.cpp:23:
/home/travis/build/probonopd/Prusa-Firmware/sketch/MarlinSerial.cpp: In function 'void __vector_51()':
MarlinSerial.cpp:68: error: redefinition of 'void __vector_51()'
   SIGNAL(USART2_RX_vect)
          ^
```

which I was not getting originally. How can I make this use Serial2?

Reference:
https://travis-ci.org/probonopd/Prusa-Firmware/builds/326280004#L619-L624

It seems like Serial2 is now somehow used already. Does this have to do with farm mode? How can I use it for communicating with the printer and for sending GCODE?

vsky279, 2018-02-25:
I had the same problem. What worked for me is to change `Firmware/MarlinSerial.cpp` on line 68 ` SIGNAL(USART2_RX_vect)` to `SIGNAL(USARTB_RX_vect)`. I think there is some kind of conflict with the 2 in the first declaration. But I was not able to got into details.

After this change I am able to compile with `#define SERIAL_PORT 2` and the printer communicates on the P3 connector.

See this commit https://github.com/vsky279/Prusa-Firmware/commit/5e65b47fb70401677b07164aa6a418c056fdcfe7.

probonopd, 2018-02-25:
Thank you very much @vsky279.

closed 2022-05-05 by Prusa-Support