prusa3d/Prusa-Firmware #391: Feature request: Consistent appearance of live-z-adjustment in the menus

3axis, 2018-01-11:
When the printer is not printing and just warming up, there is no way to adjust z so if you have just peeled off a failed print and you want to try again with fractional decrease in z gap you have to wait until the printer has actually started printing to make the change. It would be a good idea if the live z was more generally available, I think it is important enough to have it always present under the "info screen" option like it is when printing. This would also be more consistent that having it there when printing but not when not printing.

TheZeroBeast, 2018-01-28:
Hi 3axis, just in case you're not aware, there is a live-Z adjust menu option under settings when printer is idle. At least until PRUSA review your feature request.

Rob

3axis, 2018-01-29:
I will try it, I still think it is important enough to have at the top level and keep it consistent.

closed 2022-05-05 by Prusa-Support