prusa3d/Prusa-Firmware #410: 3.1.0 on MK2S repeatedly losing serial connection to Octoprint

tvi675, 2018-01-16:
Firmware version 3.1.0 on MK2S causes frequent serial disconnects from Octoprint over USB. Octoprint issue at https://github.com/foosel/OctoPrint/issues/2363 which, following investigation, seems to be an issue with this version of firmware. Rolling back to firmware 3.0.12 resolved serial disconnects.

hoegge, 2018-02-19:
Please get 3.1.1 released for MK2s also ASAP - there are many many users having this issue

halfluck, 2018-05-13:
Any development on this? I have a MK2 using 3.1.0 firmware & suffer from the octoprint disconnects.

andreiavram, 2019-04-13:
Is there anything we can help with (like testing) so a fix can be released? :-s

halfluck, 2019-04-13:
Octoprint now has a workaround for this by sending fake OK signals. also there is build 3.1.0 R10 that includes a fix works great!

https://github.com/thess/Prusa-Firmware

3d-gussner, 2019-04-15:
The latest MK2 hex file provided by Prusa is from November 2017. But in December 2017 a communication fix has been pulled into the MK2 branch. If you don't want to compile your own firmware from source code you can try my release https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.1.0-001-HFM . If you do, make sure that you read the release notes as it expects a filament Run out sensor. Mechanical and optical (like the new MK3s) sensors are supported in this "old" firmware.

closed 2023-08-28 by github-actions[bot]