prusa3d/Prusa-Firmware #414: MK3 Statistics are invalid. all firmwares including 143

cnlson, 2018-01-17:
In my case it appears that both the time and amount of filament are invalid. for others it was just the filament.

[image]

forum threads 
https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/crazy-statistics-my-printer-thinks-i-ve-printed-75-t13570.html

https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/statistics-error--t13432.html

camlesom, 2018-01-20:
Have you tried the new [3.1.1 RC5 release](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.1.1-RC5)? It says in the release notes that they fixed the failed stats.

cnlson, 2018-01-20:
i'll check it. likely this can be closed

closed 2018-01-20 by cnlson