prusa3d/Prusa-Firmware #417: Filament Unload Blob and Stuck at PTFE on the top

murathanaraz, 2018-01-18:
When unloading, there is always a blob on the filament and it stucks ath the upper PTFE tube. 

If you extrude some filament before unloading you don't get that blob... (solution by Ocie Ward)

cnlson, 2018-01-20:
I have tried this consistently since I viewed this and it has resolved all unloading issues. The filament never gets caught at the top it just comes right out with no problem

sylphio, 2018-01-26:
Extruding before unloading work for me too (thank for the suggestion).

workinghard, 2018-05-03:
Experienced the same issue. It happens with higher temp or if nozzle ran for couple minutes hot. 
Extrude a little bit and then unloading works reliable on MK3. Would be great if unload function would be enhanced!

cnlson, 2018-06-08:
@MarekC3D can you please incorporate this into the unload filament routine? extrude 15-25 mm and then immediately unload. This gets rid of the existing melted filament and makes unload extraction perfect every time.

neslekkim, 2019-05-24:
Would be nice if this could be incorporated, make the unload function extrude a bit before unloading to avoid stuck filament. Not so obvious on PLA, but a lot on say ASA.

closed 2023-10-30 by github-actions[bot]