prusa3d/Prusa-Firmware #422: Don't preheat bed for filament Unload

SukMeWot, 2018-01-21:
The current filament unload process should not ask for preheating, instead the menu should ask which fliament you are unloading and set the temperature, do the preheat of the hot end only, then eject, then cool. It should not be incumbent on the operator to select preheat, then unload, then cooldown, and it certainly shouldn't be heating the bed.

Area5142, 2018-01-23:
The preheat menu could be shown to select filament type (temperature) and only set temperature for the hot end. After unload set temperature to 0.

SukMeWot, 2018-01-31:
It does but it heats the bed too - MK3

davidhbrown, 2019-03-14:
This is indeed idiotic. I'm unloading filament because the print is done. Heating the bed just wastes energy. (Could/sould be labeled with "Firmware 3.6.")

GreenShadowMaker, 2019-04-21:
Definitely unnecessarily heating the bed with filament unload on with 3.7.0   Not sure if it was happening with 3.5.3. (I don't remember it, or I was unloading before post print cooldown had happened)

GreenShadowMaker, 2019-04-21:
looks related to  https://github.com/prusa3d/Prusa-Firmware/issues/1688

dstulken, 2022-01-06:
Any updates on this?

I noticed that the original post (Jan 2018) predates the release of the Prusa Mini (Oct 2019).     One of the best features of the Mini is its small size and portability, and because of this, unloading the filament before moving/transporting the printer is an important part of its usage.

Having the bed heat up for this is completely unnecessary... and having both the hotend and bed stay at their temps after the operation was a surprise.  I didn't even know there was a cooldown option until I found this issue request - so I suspect many others aren't cooling their machines down properly either after doing filament unloads...

SukMeWot, 2022-01-06:
This has been a bug since I've owned a Prusa - well not a bug but a piece of poor coding that has been present since slic3r from which this slicer is derived.

Prusa-Support, 2022-05-19:
> [oderbang](https://github.com/oderbang) commented [on Jan 22, 2021](https://github.com/prusa3d/Prusa-Firmware/issues/2985#issue-792172891) • 
> Printer type - MK3
> 
> Is your feature request related to a problem? Please describe.
> When selecting "unload filament" when the bed is cold the printer pre-heats the build plate. this is un necessary if you are just removing filament.
> 
> Describe the solution you'd like
> have a toggle option in the setting to enable/disable build plate pre-heat when using filament unload.

Prusa-Support, 2022-05-19:
> [thehumanfallacy](https://github.com/thehumanfallacy) commented [on Jan 16, 2021](https://github.com/prusa3d/Prusa-Firmware/issues/2869#issuecomment-761498038)
> This makes sense as I also usually want to print after loading. But maybe it could also prompt and ask you if you want to heat the bed? That should be a quick user interaction if the default value of Yes is highlighted. Just a thought.

teufelhunden0311, 2022-06-12:
they should add this

SukMeWot, 2022-06-12:
I wouldn't hold your breath if I were you. There have been countless generations where this could have been done and the filament change process is just as defective as ever. I don't believe you need an option, the bed will be heated as part of the print job. It is totally inappropriate to heat the bed for filament change, where changes are done during print the bed should maintain whatever temperature it is at.

teufelhunden0311, 2022-06-12:
> I wouldn't hold your breath if I were you. There have been countless generations where this could have been done and the filament change process is just as defective as ever. I don't believe you [...]

I know. I just figured I would comment to show my interest in it. It's weird too because a good many people defended how it is now. It seems to me that they designed this printer like it was going to be exclusively used in print farms where they never stop printing. It would be a simple toggle in the filament change submenu (preheat bed = on or off) it's not something crazy.

Prusa-Support, 2022-06-14:
Hello! Thank you for the comments, I passed this on to our devs - no promises it will be implemented yet at this time, but in case of more info, we'll update in this thread. 

Alessandro Pantaleo
Prusa Research

ThomasCZ, 2022-09-09:
Please fix this issue, it is waste of electricity.

arekm, 2022-09-09:
Shouldn't this be enough?

```
diff --git a/Firmware/ultralcd.cpp b/Firmware/ultralcd.cpp
index db53220e..33c4d578 100755
--- a/Firmware/ultralcd.cpp
+++ b/Firmware/ultralcd.cpp
@@ -1873,11 +1873,6 @@ void mFilamentItem(uint16_t nTemp, uint16_t nTempBed)
     static int nTargetBedOld;
     uint8_t nLevel;

-    nTargetOld = target_temperature[0];
-    nTargetBedOld = target_temperature_bed;
-    setTargetHotend0((float )nTemp);
-    setTargetBed((float) nTempBed);
-
     {
         const FilamentAction action = eFilamentAction;
         if (action == FilamentAction::Preheat || action == FilamentAction::Lay1Cal)
@@ -1897,6 +1892,19 @@ void mFilamentItem(uint16_t nTemp, uint16_t nTempBed)
         }
     }

+    nTargetOld = target_temperature[0];
+    nTargetBedOld = target_temperature_bed;
+
+    setTargetHotend0((float )nTemp);
+    switch (eFilamentAction)
+    {
+        case FilamentAction::Load:
+        case FilamentAction::UnLoad:
+            break;
+        default:
+            setTargetBed((float) nTempBed);
+    }
+
     lcd_timeoutToStatus.stop();

     if (current_temperature[0] > (target_temperature[0] * 0.95))
```

Prusa-Support, 2022-10-19:
Hello, thank you for the comments, @arekm sadly I wouldn't be able to fully confirm the code from here but we are reviewing it with devs 🙂 

Alessandro Pantaleo
Prusa Research

dstulken, 2023-08-21:
> This issue has been flagged as stale because it has been open for 60 days with no activity. The issue will be closed in 7 days unless someone removes the "stale" label or adds a comment.

Not stale.  Very much still an issue.

gudnimg, 2023-08-21:
@dstulken is it not resolved with 3.13.0? We merged https://github.com/prusa3d/Prusa-Firmware/pull/2335 There is a toggle menu item now at the bottom of Settings to enable this.

Prusa-Support, 2023-08-22:
Indeed this issue should be closed but technically not because stale. 🙂 

The PR https://github.com/prusa3d/Prusa-Firmware/pull/2335 that resolves this issue was merged in the firmware and integrated into the new [FW 3.13.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.0) release.

Thank you all for your patience and collaboration and please allow us to close this issue now.

Michele Moramarco
Prusa Research

closed 2023-08-22 by Prusa-Support

dstulken, 2023-08-22:
It's not resolved on my end...  Prusa Mini running the latest firmware (4.4.1).

My apologies, I hadn't realized at the time of my other posts that the firmwares for different printers have separate GitHub projects, but had keyed in on this one as it seemed to be the oldest "upstream" issue that others were being folded into.   

So I'll ask the relevant question:  Since this has been fixed here for a portion of the Prusa lineup, and the solution is fresh in the devs mind, could this also be ported to the Mini firmware?

Prusa-Support, 2023-08-22:
It is safe to assume it will be ported to Mini because this is how the MK4 works.
A portion of the MINI's base code will be developed along with the MK4 firmware.
Please refer to https://github.com/prusa3d/Prusa-Firmware-Buddy repository.

Michele Moramarco
Prusa Research