prusa3d/Prusa-Firmware #423: Fliament load - hot end too close to the bed

SukMeWot, 2018-01-21:
If the user selects load filament the hot end should be moved either to the top of the Z axis or certainly half way.

If a user loads filament too close to the bed it blobs on the bed and is near impossible to see when the question is asked re the colour of the extrusion.

3d-gussner, 2018-01-22:
Which printer MK2 or MK3? Which firmware version do you have?

The Filament change routine has been modified to raise the hotend if is too low.

- After turning the printer on the filament load can (and should able to) be done without homing and then it doesn't know its position, so what is the solution here? I think in that case the user responsible to avoid that by manually rising the hotend.

- On the other side if the printer 'knows' its position it should do the same as the change filament routine: raising the hotend if too low.

Area5142, 2018-01-23:
It is very easy to move the extruder to a "safe" location for filament load - just press and keep the knob in and you will enter the z-axis move mode. Lift extruder and press again. Load filament.

SukMeWot, 2018-01-31:
It is a MK3. First load was way too low, asked me about filament and it was so low you couldn't see it, not tried it since but I have updated.

And yes I could move it to a safe location but I shouldn't have to - it should all be part of the load / unload sequence.

westranm, 2018-06-07:
I'm running 3.2.0-555 firmware and Prusa i3 MK3 is my first 3D printer.  I was certainly expecting that something as simple as this would have been part of the refinement process on this printer by now.  Please update so that choosing to load a filament puts the extruder in a convenient position to load and clean off the hot end.

Prusa-Support, 2022-05-12:
Hello! Sorry for the late response on this thread. I am closing the issue for now as this has been addressed in the meantime: prior to a load, print head would indeed raise with current FW. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-12 by Prusa-Support