prusa3d/Prusa-Firmware #424: Feature request - preheat should not head hotend to full temp to avoid filament leaking

cnlson, 2018-01-22:
ABS and higher temp preheats have the hotend at full temp in a minute and 9 minutes for the bed which leads to filament leaking from the nozzle for 5-6 minutes before the printer is ready to print.

abs should preheat to 190/100 so that when you start a print it takes a minute or so to finish the preheat on the nozzle which limits the amount of leakage to deal with.

bubnikv, 2018-01-22:
This certainly makes sense. But do you want this logic to be implemented
inside the firmware, or should it go into the Slic3r start G-code sequence?
I would think the 2nd makes more sense.

Area5142, 2018-01-23:
The easiest fix is to place it in gcode.

1) Start heating hot end to 190 and hot bed to 100
2) Wait for hot bed to react 100
3) Heat hot end to extruding temperature
4) Wait for hot end to reach extruding temperature

If hard coded down in the firmware, it is very difficult to change and other users may want to have a different heat sequence for another filament.

cnlson, 2018-01-23:
i was thinking in general that it would be good to have in the firmware a limit that speeds the heating but does not get it to the point of drooling. but i do see your point.

cnlson, 2018-02-02:
On a second thought I really don't think this is a bad feature request. The point of preheat is not to get the filament drooling all over the bed it is to get the bed up to temperature since the bed takes so much longer  to warm up. 

For example with ABS the hotend is preheated at 90 seconds the bed, 6-7 minutes. I could see delaying the hot end as well but if it is just below the drooling point then within 5 seconds it is ready to print since going 190 to 255 is roughly that long

cnlson, 2018-02-02:
@Area5142 can't do that in prusa control, no options for it.

Area5142, 2018-02-02:
@cnlson Prusa Control can generate the necessary gcode, bases on selected filament. That would be better than the firmware handling all filaments equal with no way of change it.

_Expert users_ can make their own gcode in Slic3r and Prusa Control users have the best heat sequence based on selected filament.

cnlson, 2018-02-02:
@Area5142 right now filaments are "all equal" for pla 215/60 for abs 255/100 for pet 240/90. all i am suggesting is that instead, they be 215/60 (since they achieve these temps nearly at the same time) 190/100 (since it takes minutes more to get to 100 than it does to get to 255) and 190/90 for the same reason.   that way if i start it and go get a drink or go to the bathroom, by the time i come back the bed is at operating temp and there isn't a foot long string of material i have to deal with before starting a print. i can just click print from sd and go

Area5142, 2018-02-03:
@cnlson I agree on the problem, but disagree on the solution to hard code a specific heat sequence in the firmware.
The most logical solution is to have the slicer (Slic3r/S3D/Prusa Control) generate the necessary g-code from the knowledge of filament type and/or selected profile.

A simple solution I found long ago was to include this _filament end g-code_:
```
; Filament-specific end gcode
G1 E-3.0 ; Retract filament 
```
When ever the print finished the filament is retracted 3mm and I will have no dripping at start of the next print. As this is filament specific, I only put it in filaments with problems.
I know there are situations where this solution will not work - it is just an example.

dalymople, 2018-12-14:
I agree that this should be handled by firmware and not in the g-code. If it is in g-code it requires every user to tune it themselves or add special cycles, whereas the firmware could calibrate itself automatically. In reality no filament should be kept hot unless it is being extruded, so the printer should minimize the time that the filament is at this temperature.

All that the firmware really needs to do is monitor the rate of rise of the bed and hot end temps during pre-heat and simply adjust the PWM output to try and make them both reach temperature at the same time.

This would be more `filament friendly` and would reduce drooling / stringing during pre-heat.

cnlson, 2018-12-14:
I've adjusted my slic3r profiles and really don't have the issue much anymore. I'm of the opinion that preheat should only heat the bed since the bed is the item that takes the longest and a separate load /unload preheat should be used to turn on the hot end. Otherwise there really is no reason to preheat filament since it takes 90 sec or less to hit 300 so what's the point.

Pathogenius, 2018-12-15:
I agree that the default g-code might not be optimum as far as the heating routine. Until I decide on the tweaks, I typically preheat the bed to 50 for PLA prints, and let it sit there while I clean the sheet and copy the stl to the sdcard etc. As cnlson points out, the extruder preheats very quickly, especially if you forgot to grease the heatbreak when you rebuilt the extruder.

cnlson, 2018-12-15:
... and copy the stl to the sdcard

you need a flashair card, send your gcode by wifi :+1: