prusa3d/Prusa-Firmware #425: Allow seperate Z offset heights for ABS and PLA

mrjohnc, 2018-01-22:
Whenever I switch from using ABS to PLA I have to manually change the Z offset because ABS needs more 'squish'. It would be really helpful to have an option in the calibration to be able to set which filament you are printing and then store an offset for each filament. This offset would then be used when printing the different materials.

clearbucketLabs, 2018-01-30:
you can use the slicer for that.  long as it takes a negative Z...

macfly1202, 2018-01-30:
Very nice idea... I have got the same problem and improve the firmware to remove thins manual Z live adjust  will be cool.

mrjohnc, 2018-01-30:
Ah, maybe this could be done in Slic3r instead then, like add a reduced offset to certain filaments?

chilternflyer, 2018-02-16:
You didn't say which machine you are using.
The Mk2 PINDA probe still isn't 100% accurate when you change the bed temp from 60 to 100c.
I.E. the detected bed height is approx' 0.15 - 0.22 higher at 100c than it is at 60c.
If you are printing on a Mk2 then you need to adjust the first layer height in your ABS profile(s) in your slicer.

mrjohnc, 2018-02-20:
@chilternflyer I'm using the MK2, yes, I know you can change this setting in Slic3r, what I'm trying to ask is if this height adjustment can be done somewhere automatically (in firmware, PrusaControl or somewhere else)

Thanks

chilternflyer, 2018-02-20:
Hi John,
setting it in firmware is pointless because the adjustment of Z height will vary from probe to probe.
My solution on my Mk2 was to move the extruder well away from the bed after it had finished printing. 
(The idea being that it stops the heat from the bed affecting the probe in between prints).
This is easily achieved by modifying your end gcode to move the hot end up and away from the bed (I.E. add something like G1 X0 Y200 Z165 to the file).

mrjohnc, 2018-02-20:
So you are suggesting the lack of squish is coming from the accuracy of the probe at different temperatures rather than the mechanical properties of ABS? 

Could your solution of moving the extruder away from the bed at the end of the print be added to the standard G Code produced by Slic3r and Prusa Control so that it would improve the accuracy of bed levelling for all filament types?

Panayiotis-git, 2018-02-20:
I'm moving the extruder away from the bed also at the beginning of the print.
Just home X, Y, Z without bed leveling and move the head at some height above the bed (100mm)
I do not have any issues with the regular materials PLA, ABS, PETG.

chilternflyer, 2018-02-20:
John: Yes and yes..
The PINDA probes are temperature sensitive so if you calibrate the printer on a bed set to 60c and then use it on a bed set to 100c, the probe will heat up and will detect the bed at a slightly different height.
So yes:- The lack of squish is caused by the probe temperature being different from PLA to ABS (or from room temperature to a hot bed).  It has nothing to do with the material.

I keep my extruder away from the bed so that the bed temperature has minimum heating effect on the probe. 
So yes:- Adding the code to move the probe away from the bed **at the end** of the print is the best way to improve bed height detection for all materials.  The other thing that will help is to let the bed cool down (below 40c) before starting your next print.
Don't forget that you do have the **live Z** adjust in the menu on the control panel of the printer.  I use this whenever I change reels or I'm rushing to get the next print started.

clearbucketLabs, 2018-02-21:
Also the nozzle is the closest and to the probe and usually you use higher temp with ABS which could also throw it off.

closed 2022-05-05 by Prusa-Support