prusa3d/Prusa-Firmware #428: Constant vertical banding

ff8jake, 2018-01-23:
All MK3 prints seem to be plagued with vertical banding. The B150 firmware dropped jerk to 10, and attempting to fix the issue I tried with 5 and still see the artifact. In the image below you can see the persistent lines, and what little ringing there is off the "Y" disappears shortly after the "Y" while the vertical artifact continues.

[image]

On my printer I exchanged the stock bearings with IGUS and this decreased slightly, but is still there. Slower speeds make this slightly less noticeable, but still there and at the same frequency (which makes it seem related to stepping function instead of vibration).

Once you see this artifact (very visible on gloss PETG prints, you can even see it on the MK3 extruder parts!), it's very easy to spot elsewhere. Nearly all images of pics posted on the forums show this artifact to some degree or another.

I am able to compile the firmware and test, if anyone with more experience has some ideas for things to try. I am at a point where I am hesitant to touch settings in firmware as I am not entirely sure what to be looking for or what the changes may do.

Thanks!

libxmike, 2018-01-27:
I can confirm that, the only way MK3 can be used is with 15mm/s external perimeter which is not acceptable for 1000 EUR printer.
Also installed IGUS on X with same effect, cant go over 15mm/s otherwise ringing and severe vertical lines.

slayer1551, 2018-01-27:
Can confirm I also have this as well as others on the Facebook groups. I've upgraded to Misumi rods and bearings which helped a bit but there is still odd ringing artefacts. Printing at 15mms minimises the issue but the MK2 can print outer perimeters at 35mms without issues so its a big step backwards regarding print time just to get the same quality print

ff8jake, 2018-01-28:
PROGRESS!!!

This is belt/smooth idler related. Saw someone with a similar issue ( https://www.duet3d.com/forum/thread.php?pid=30141#p30141 ) who was able to fix this problem by putting a twist in their belt so the smooth side of the belt was on the idler. I tried this and got the following results, both cubes done with the same settings (.2 layers, 45mm/s ext perimeters):

[image]

Thus far this is the only modification I have made that caused an effect. I believe the ripples that are left behind now are legitimate ringing/ghosting caused by jerk/accel settings. To highlight this on the right hand photo, I took the photo so the reflection was on the smooth spot that is NOT there on the left hand photo. The left hand photo lines are consistent in that area.

That said, I am about to try to find a suitable GT2 tooth'd idler to print. Also from that topic was the following:

> With the gates curvilinear profile (GT, GT2, GT3, etc), it's acceptable to run the teeth side against a smooth idler if it has a diameter equal to or greater than the effective diameter of a 40 tooth pulley of the same tooth pitch.

> With 2 mm pitch belt, this equates to a smooth pulley with about 26 mm diameter.[/quote]

That cube is 20mm, so of course it doesn't fit that spec! Also is anyone else willing to give this twist a or tooth'd idler idea a shot?  :)

chilternflyer, 2018-02-16:
I also have the same ringing issue. It seems to happen more on the x axis. Even though the belt is tight enough it still seems to resonate during certain moves (lower part of the belt on the motor side).
I'm also wondering if the idler pulley is misaligned (horizontally) because the belt seems to have a bias to rub on front side of the idler pulley. 
Does anyone else have similar mechanical characteristics on their Mk3?

messerrt, 2018-06-01:
I'm having the same issue. I've been using my MK3 kit printer for about a month.  Sense the first Prusa logo print from the SD I have had the vertical banding.  I thought it might be because of the X idler scrubbing but it only scrubs each revolution. Its the same on all surfaces. I'm currently printing some Hexagon Terrain pieces for a fried and every side is identical. I'm a mechanical Engineer and I know that my machine was assembled per the instructions and experience. My printer is useless until this is fixed. I sell my designs to the gaming community. I invested in this machine because I expected quality prints. I agree with chlternflyer, the idler pulley is a poor design. Plus the only way to access it is to remove the right Z column. 
[image]

koenkooi, 2018-06-02:
Is this the same issue https://github.com/prusa3d/Prusa-Firmware/pull/771 tries to solve?

messerrt, 2018-06-02:
Yes, That sounds exactly like what I'm getting.

koenkooi, 2018-06-04:
Disabling microstep interpolation in X,Y and E seems to have decreased this effect on my printer. Only 3 test prints so far, so it could be placebo effect.

```
--- a/Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h
+++ b/Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h
@@ -195,9 +195,9 @@
 #define TMC2130_USTEPS_XY   32        // microstep resolution for XY axes
 #define TMC2130_USTEPS_Z    16        // microstep resolution for Z axis
 #define TMC2130_USTEPS_E    32        // microstep resolution for E axis
-#define TMC2130_INTPOL_XY   1         // extrapolate 256 for XY axes
+#define TMC2130_INTPOL_XY   0         // extrapolate 256 for XY axes
 #define TMC2130_INTPOL_Z    1         // extrapolate 256 for Z axis
-#define TMC2130_INTPOL_E    1         // extrapolate 256 for E axis
+#define TMC2130_INTPOL_E    0         // extrapolate 256 for E axis
```

As you can see I raised microstepping on X and Y as well, but that didn't seem to have any effect on the ripples.

koenkooi, 2018-06-04:
This is a picture of the banding I had *before* doing any firmware changes:
[image]

https://www.thingiverse.com/thing:2835132 at 50% scale (or rather 1270x scale since the author confused inches with millimeters during export to STL).

ff8jake, 2018-06-04:
I honestly forgot I had this issue open. There's slight vertical scratching that seems to be caused by linearity correction possibly, but I found this posted on the forums awhile back:

[image]

The user who posted this printed it half way with the stock rubber feet, then printed the rest with rigid feet. Since then I have designed these https://www.thingiverse.com/thing:2802540 and have them on 1" felt pads for rigidity. The printer will be a bit louder.

devilhunter84, 2018-07-30:
@koenkooi 

What's the status on this?
Did disabling microstep interpolation in X,Y fix this for good, or do you still have the artifacts?

koenkooi, 2018-07-30:
Artefacts are still present :(

musyne, 2018-10-09:
Same thing here. On MK2 and on MK3. Not so visible with functional pieces but very apparent on "organic" shape. Image below is a print on the MK3 with the 0.10 detail preset (only changes are 3 walls and support).
[image]

Crunch69, 2018-10-19:
has there been any solve to this issue?  I have tired slowing things down and speedi ng things up and the vertical lines still show up

BapPerraud, 2018-11-27:
Hello everyone !

Have anyone found a solution to this problem ? I search on internet for a long time now, for Something explaning these verticals lines. Unless this problem, my prints could be near perfect.. I see a lot of topics there and there but never saw someone who resolved this problem.

[https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/vertical-lines-on-y-t23047.html](There)  [https://shop.prusa3d.com/forum/-f6/vertical-lines-t1279.html](There) Etc..

So I'm facing the same situation than a lot of people: vertical banding on my prints. Saddly, I could not determine if this is a hardware problem or software/configuration problem :(
Note: I'm running a Prusa i3 MK2.

[image]

[image]

I still hope that someday someone will investigate this issue that a lot of mk2/mk3 users seem to face.
Have a nice day !

austinvojta, 2019-01-23:
+1 to the list of people frustrated with this MK3 issue. My old MK2S prints were buttery smooth. Only happens along the y-axis for me, x-axis-onky movemenets are totally smooth. Belt tension certainly not the issue. I may try replacing the stepper motor or throwing on a TL smoother just for the hell of it.

[image]

Allarabie, 2019-04-10:
_TLDR: Adjusting motor current fixed it for me. Try increasing X motor current and watch as the pattern changes.. I was able to dial mine in and get rid of almost all the vertical banding._

This may or may not be apply to your case.
I recently designed and build my own custom large format printer (310x310x350).
And was struggling with this exact same vertical banding. I came to find out that is is caused by m X stepper motor.  I tried swapping motors, gears, etc.. nothing helped. 
After many MANY google searches, I found a post stating that; if your stepper's are not getting enough current it may not me able to hold a micro-step.  This causes it to skip, or jump to the next full or half step (which ever it has enough current to hold).
Im using an Einsy Rambo (TMC2130), so I can change motor current with gcode. What I did was print a test tower 20x20x50tall, and would change the current every 10mm or so. I would watch the pattern change and after a while, I found the perfect current and almost all the vertical banding is gone.

Anyway, hopes this helps someone.

austinvojta, 2019-04-10:
I am so excited to try this. Thank you for sharing! Will definitely report
back with results when I can.

Allarabie, 2019-04-10:
I really hope it helps, I've been trying struggling with this for almost 2 months. 
An other thing I had to be adjust because of the current, was Sensorless homing sensitivity for my X axis. You may or may not need to..

stahlfabrik, 2019-04-10:
I will try that too! How high did you go with currents? What step size do you propose?

Allarabie, 2019-04-10:
I wont claim to be an expert at any of this.  And keep in mind this is a DIY printer made mostly of spare parts.. The steppers Im using were the originals from my 3 year old HicTop, so not exactly quality.

After quite a bit of testing (and Im still not 100% happy), I ended up setting Current to X780 Y700,  Microsteps for X and Y to 32, and Steps/MM to 161.24. 
Both X and Y steps/mm are the same and their both accurate to about 0.1mm

BapPerraud, 2019-04-11:
Hello Allarabie and thank you so much for sharing this. I will check and try this as soon as possible and come back here to share the results.

messerrt, 2019-04-11:
Help a noob. What was the gcode you enterd?

Allarabie, 2019-04-11:
Well again, Im not doing this with Prusa firmware (or even a Prusa machine), this is stock Marlin 1.1.9.
Anyway, the current command is M906.
So for example:   M906 X740 Y700 Z400 T0 E500
This will set the current for your X motor to 740, Y to 700, Z to 400, and Tool 0 Extruder to 500
Ans you can do them 1 at a time. 
So if you only want to change X to 500, you would send: M906 X500

I believe that Prusa's stealth mode cuts the current to the motors to make them quieter, I wonder if that would be related.. 
Are any of you getting these results in Stealth mode?

leptun, 2019-04-11:
On prusa firmware you can't change the current from gcode by default. You have to recompile the firmware with "#define TMC2130_SERVICE_CODES_M910_M918" uncommented in configuration_prusa.h (This is optional though and just for testing as you can set the values directly like in the picture).
With the compiled firmware you can do the following from gcode (default values supplied):
- **Set hold currents : M911 X16 Y20 Z35 E30**
- **Set running currents M912 X16 Y20 Z35 E30**
- Print currents : M913
- Set normal mode : M914
- Set stealth mode : M915
- Set/Print crash detection sensitivity M916 X3 Y3 Z4 E3
- Set pwm_ampl : M917
- Set pwm_grad : M918

This changes do not save on reset though. To save them you need to change the values in the same file
[image]
The values are not in mAmps RMS like in marlin. You can just make the default values a bit higher for X and Y

stahlfabrik, 2019-04-11:
I prepared a "Before Layer Change"gcode that I will try with a 5cm cube after my current print. Running Marlin 2 on my MK3. 

Cannot tell yet if that will work. X282 and Y348 are Prusa default values in mA!

{if layer_z==0.2}
; Power level 1
M906 X282 Y348
{elsif layer_z==5.2}
; Power level 2
M906 X302 Y368
{elsif layer_z==10.2}
; Power level 3
M906 X322 Y388
{elsif layer_z==15.2}
; Power level 4
M906 X342 Y408
{elsif layer_z==20.2}
; Power level 5
M906 X362 Y428
{elsif layer_z==25.2}
; Power level 6
M906 X382 Y448
{elsif layer_z==30.2}
; Power level 7
M906 X402 Y468
{elsif layer_z==35.2}
; Power level 8
M906 X422 Y488
{elsif layer_z==40.2}
; Power level 9
M906 X442 Y508
{elsif layer_z==45.2}
; Power level 10
M906 X462 Y528
{endif}

Spectrewiz, 2019-04-21:
Wanted to share my initial results from testing increasing the TMC2130 currents with the M911 and M912 commands in the Prusa firmware. Setting X to 18 and Y to 20 nets noticeable improvement with the vertical bands in my small test print which seems promising, but be careful setting them too high. When I set X to 22 and Y to 26, my test print crashed - I didn't change any other settings. I'm going to try figuring out why the crash occurred, but for now I'm going to print two cubes, one with the default currents and the other with the slightly higher currents. I'll post my results

Spectrewiz, 2019-04-21:
Unfortunately the test cubes that I printed both look the same even with the modified current values. The original test print looked pretty clean (before it crashed) but it was rather thin. I'll keep testing. Really hoping that we can find a fix for this.
[image]

austinvojta, 2019-04-22:
Thank you for testing, Nelson. Wish I could contribute but my MK3 is not
operational at the moment.

nihlton, 2019-04-26:
Does anyone know if Prusa folks have acknowledged / addressed this issue?  are they looking at it?  Working on a fix?

Upgraded a few months ago from a PrintrBot simple metal and i've just been all around so pleased - except for this one GLARING print quality issue :T

HMPoweredMan, 2019-04-29:
> Does anyone know if Prusa folks have acknowledged / addressed this issue? are they looking at it? Working on a fix?
> 
> Upgraded a few months ago from a PrintrBot simple metal and i've just been all around so pleased - except for this one GLARING print quality issue :T

I also upgraded from a simple metal a few weeks ago. This is the only issue. I think the XYZ linearity settings may be of use from what I've seen in other threads though there is no official documentation.

The printed parts from prusa even have this so I doubt it's something with a solution as of yet.

Spectrewiz, 2019-05-05:
I've done many tests now with altered current values, hysteresis parameters, and microstep resolutions. Unfortunately all of them still have the vertical banding artifacts present. Here is a list of things I've tried:

### Tuning `hstrt` and `hend` with gcode
I created gcode commands M919 and M920 to modify hstrt and hend on the fly so that I could test different values at mm increments throughout test prints ([see the code here](https://github.com/Spectrewiz/Prusa-Firmware/blob/ce03c8b17809c16fdc89aa84b77773e08e3b5da5/Firmware/Marlin_main.cpp#L6887)). The reason for doing this was to brute force a solution to motor currents zero-crossing from the insights in #771. The idea was that if zero-crossing was the cause of the vertical banding artifacts, trying out many combinations of the hysteresis parameters throughout multiple intervals in test prints would eventually lead to the correct combination and the vertical lines would cease. Page 50 in [the TMC2130 datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2130_datasheet.pdf) details the possible values for `hstrt` and `hend` (0..7 and 0..15 respectively) and explains how they work together.

I did many tests and no pair of parameters seemed to make noticeable differences in the vertical lines. This implies that the issue lies elsewhere or that the issue may be a combination of the hysteresis settings and other problems.

### Tuning `currents_h` and `currents_r` with gcode
The gcode commands M911 and M912 modify the registers in the TMC2130 that control the holding current and running current of the motors. The idea was that if improper current levels were the cause for the artifacts, trying out different levels throughout intervals in test prints would eventually lead to the correct level and the vertical banding would cease. 

In my tests, I only modified the current for the X and Y axis as that's what @Allarabie mentioned fixed his banding issues in his printer. Throughout the tests I kept the holding current and running current the same (that is `current_h = current_r`) and tried currents from the default parameters (X16 Y20) up to X35 and Y40. I had to modify the stallguard sensitivity with M916 in order to allow the currents to get this high, as the printer would continuously crash if the setting was too low. The only noticeable difference that I could tell was that higher currents made corners crisper and the printer louder, but did not seem to reduce the vertical lines. 

### Using stealth mode instead of normal mode
Stealth mode and normal mode can be switched with M914 and M915, and it was actually the first time I used stealth mode on any of my prints. I have to say, the quietness of the printer is very impressive when stealth mode is enabled. Anyways, stealth mode enables StealthChop in the TMC2130 which is different from the SpreadCycle chopper used in normal mode (see page 39 in the datasheet). This means that the hysteresis tuning and current control that I did in the above tests wouldn't apply here. So the only thing I tested here was to turn from normal mode to stealth mode halfway through a test print and see if that made any differences in vertical banding. Unfortunately, the vertical banding was still present, but it did seem somewhat reduced.

### Conclusion
From my tests, I believe the issue that causes the vertical banding lies elsewhere. From some research, I was able to pinpoint some solutions that others have tried for other printers, such as inducing a half twist to the belts so that the belt teeth don't run over the smooth idler bearings. Apparently the teeth compress and decompress when running over the smooth idler which can be a cause of the vertical lines in the prints. We have smooth idlers on our Prusa printers, so this may be something to try out. We could also try getting or printing grooved bearings and see if that helps. [Source](https://www.reddit.com/r/3Dprinting/comments/7d1pqe/vertical_surface_ripples_banding_waves_solved/).

The other thing I noticed was that there seems to be a good deal of vibrations in the frame from the motors running during the prints. The extruder head also seems like it can be "wobbled" back and forth parallel to the Y axis when pressing down from the top, maybe that can have something to do with it? I don't know much about resonance or how vibrations would create the perfectly spaced vertical lines, but it was just something that I noticed. I created a gif to demonstrate what I'm talking about, forgive the low quality it was the only way I was able to email it to myself.

[image]

I would love it if some others would chip in and double check my tests / try their own tests and post back with their results. I am curious to hear what came of @stahlfabrik's current level test to see if he got different results than mine. Thanks for reading, please let me know if you have any ideas or feedback.

HMPoweredMan, 2019-05-05:
> I've done many tests now with altered current values, hysteresis parameters, and microstep resolutions. Unfortunately all of them still have the vertical banding artifacts present. Here is a list of things I've tried:

I have a few things I plan to try. You mentioned smooth idlers. I ordered some replacement toothed idlers that I plan on installing after modifying the X axis. I plan on using this X axis and carriage:

https://www.thingiverse.com/thing:3569345
https://www.thingiverse.com/thing:3559732

These will accomodate the bondtech 3:1 extruder and the gates belts and pulleys I want to use.
https://www.bondtech.se/en/product/prusa-i3-mk2-5-mk3-extruder-upgrade/
https://e3d-online.com/blog/2018/11/29/gates-belts-pulleys-and-idlers-now-available/
These should make things a little more rigid.

Finally I am replacing the X and Y stepper motors with moons motors as the stock motors seem to be the biggest offender that can't be tuned out. According to this thread I am upgrading to the 1.8 degree moons steppers.
https://prusacommunity.com/to-the-moons-and-back/
https://www.amazon.com/gp/product/B071F163WT/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

I'll let you know the results of this but it may be a while as I am taking a vacation

nihlton, 2019-05-05:
So this isn't a great solution, but I've gotten rid of the banding by slowing the perimeter shell to 20% of the base print speed.

Also increased the base print speed a little so the interior shells and infill print faster to compensate a bit.

Down sides: prints take a bit longer, and simplify3d's estimates are now *wildly* innacurate.

BapPerraud, 2019-05-05:
Hello and thank you for reporting back.

I tried many things by myself but couldn't eliminate the vertical banding (changing steppers current did not help).
At this point, after many research across the web, I never found any responses from prusa to this, it doesn't seems like they care or maybe they know something they don't want us to know. I meet them on a maker faire convention and show them a printed object showing this vertical band problem, and they redirected me to this: https://help.prusa3d.com/article/p2xdpg0ul8-extruder-linearity-correction-calibration, Which seems to address a different problem. (And couldn't apply to an i3 mk2)

So unfortunately, after many years of experimentations by dozens of makers, this problem is not yet resolved :(

Panayiotis-git, 2019-05-05:
Have you checked the following?
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../fine-vertical-artifacts-trinamic-chop-tuning-any-effect/

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../stepper-motor-upgrades-to-eliminate-vfa-s-vertical-fine-artifacts/

https://github.com/guykuo/Prusa-Firmware/tree/0.9-Degree-Stepper-Support

Elessar-lad, 2019-08-06:
Hi!

I'll add my two cents here. '

I have the Bigtree tech SKR V1.3 (not the same as the Prusa Mk3) and the TMC2130 in all the axis + extruder, and I've been experiencing that z banding. I'm using the Ender 3 by the way. 

I had this issue come up about 2 months ago, and for the life of me I couldn't figure out, after hours of troubleshooting I decided to download Marlin 2.0 again and redo all my setting and flashed the new firmware, that did the trick... or did it? It did fix it at that time, but over the weekend I had the same issue yet again after replacing the extruder, I kept a copy of that firmware, flashed it and not surprisingly the issue was still there. I had the hybrid threshold disabled and also reduced the current of the motors, I increased the current of each axis one at a time and printed something, nothing changed really. 

I did the following, I ran the PID tuning for the hot end and bed, and also enabled the hybrid threshold. Printed something and looked great. I left it running a 4.5 hour print and came beautifully, and so far, seems that fixed it! I printed something else and before leaving to work looked amazing after a 1 hour of being printing. 

How much this will apply to the Prusa printers I'm not quite sure, but I thought on sharing this. 

Regards,

VendicarKahn, 2019-10-04:
I have an Anet-A8 and it suffers from the same banding.

In my case the spacing of the bands on the print is the same as the spacing of the drive teeth on the belt.

You can hold the belt against the model and the teeth will line up with the bands.

Elessar-lad, 2019-12-26:
To add more to this topic. 

I've been affected by this issue on my non-Prusa printer. I had it since I move to a 32-bit board with TMC 2130 v2.1 by BigTreeTech. 

After weeks and countless hours invested trying to figure the issue mechanically, I decided to move to the firmware. I tried many, many things there, it improved a little bit if anything. I was about to give up and I remember that I never checked what was the VREF on the drivers, just plug them on the board and changed the RMS current via software, nonetheless, to my knowledge there's no way to know via Gcode what the VREF is, so I couldn't really know how much voltage the motors were getting. 

I proceeded to check the VREF and it was 1V on each of the drivers, the motors on the Ender 3 are rated for 0.84V and 0.9V. I have been overpowering my motors since I switched to the new hardware, yikes! I adjusted the VREF the best I could knowing the RMS current (using teaching tech's YouTube video). I was so happy to see the problem go away after so much time invested. 

I don't know if this will be the case for the Prusa printers but honestly, it is worth checking the voltage. 

Before and after picture. [https://imgur.com/gallery/GBG7ANa](url)

sedlacekdavid, 2020-02-25:
Having the same issue. TMC2209 ender 3. The only thing that can be changed is steppers current through menu.
There is no physical potentiomener. Unfortunatelly changing the current (.6 or .81) has not worked.

GibsDev, 2020-03-26:
I am experiencing this issue as well with Marlin and an MK3 clone, but only on my Y axis. I had a theory that it may be smooth idler pulley, as I have a toothed one for X. After a quick google search, I found this:
https://imgur.com/a/kb2nE
I'm going to try the half twist first, but I might opt to get a 20t idler pulley instead. Alternatively I could modify my Y bed anchor and idler to work with another 16t idler I already have.

GibsDev, 2020-03-26:
Seems like the half twist only ever so slightly dampened the vertical ringing on the Y. I would still like to try a toothed pulley to see if anything changes.

austinvojta, 2020-03-26:
Hi guys -- this has been effectively solved. The stock motors have linearity issues. Check out this forum post from Guy Kuo who did very extensive testing with impressive results. Personally I'm going to grab some Moons' 0.9 stepper motors for X and Y soon to get rid of my banding (of VFAs, as the forums decided to call this effect). https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../stepper-motor-upgrades-to-eliminate-vfa-s-vertical-fine-artifacts/

closed 2023-09-05 by github-actions[bot]