prusa3d/Prusa-Firmware #435: Turn off G-Code Text description

ArchwayJourneys, 2018-01-24:
Has anyone found a setting to turn off the window that pops up when your cursor is over text in any of the G-code boxes?
This is a new issue for me after updating my operating system and loosing previous Slic3r settings.
I had never encountered this annoying pop up window before but considering the length of some of my code it makes editing the code inside the slic3r software impossible.

If anyone has any suggestions on how to turn off this annoying dialog box it would be greatly appreciated.

[image]

Panayiotis-git, 2018-01-24:
This has always been an annoying feature of slic3r.

ArchwayJourneys, 2018-01-24:
I have no idea what I was using before but lately it is so obnoxious I have to write all my code in a separate text file since the top left corner is centered over your cursor blocking out any text your typing or editing.
I switched to Prusa slicer 1.38.5 from 1.36 when I noticed all my settings were erased or lost after my windows 10 operating system update.  
Another abnormality i noticed is the gcode text organization. both prusa 1.36 and 1.38 wrap the text in a jumble(see attached). the strange thing is the version of slic3r i was using didnt wrap the gcode as i had done edits with code text files in the past.
[image]
 
I found the (non-prusa) Slic3r to keep the g-code in the correct tabulated format. 
[image]

If only i could activate a slicer button in the (non-prusa) Slic3r software to re-slice models that stopped generating the slice in preview after adjusting the slice settings.

bubnikv, 2018-01-25:
> I switched to Prusa slicer 1.38.5 from 1.36 when I noticed all my
settings were erased or lost after my windows 10 operating system update.

We moved the Slic3r user settings from a Slic3r directory to Slic3rPE
directory.

> Another abnormality i noticed is the gcode text organization. both prusa
1.36 and 1.38 wrap the text in a jumble(see attached). the strange thing is
the version of slic3r i was using didnt wrap the gcode as i had done edits
with code text files in the past.

The Windows notepad does not recognize Unix end of line separators
correctly. You can edit the G-code with a Windows WordPad, or better use a
free https://notepad-plus-plus.org/ or something similar.

ArchwayJourneys, 2018-01-25:
Thanks bubnikv Notepad++ will be a great workaround.
I wonder what is different where the Slicer now wraps text and (non-prusa) as well as my older version didn't wrap the text.    

Any suggestions on eliminating the g-code pop up text description i mentioned in the first post?

bubnikv, 2018-01-26:
> I wonder what is different where the Slicer now wraps text and (non-prusa) as well as my older version didn't wrap the text.

It is a difference in the end of line definition on Windows, Linux and OSX. Windows traditionally uses two characters as a newline separator: Carriage Return (CR) and New Line (NL), so you will find CR NL at the end of a line generated by the old Slic3r. Our new slic3r writes just the NL on Windows, LInux and OSX. The Unix convention (just NL) is the usual convention on Unix/Linux and OSX, and most of the text editors support it even on Windows, the old Notepad being an exception.

> Any suggestions on eliminating the g-code pop up text description i mentioned in the first post?

It is a known annoyance, and yes, I am annoyed by it as well. We have @YuSanka working on this part of code right now, so she will hopefully improve it.

ArchwayJourneys, 2018-01-26:
Great response thank you!

closed 2018-01-26 by ArchwayJourneys

YuSanka, 2018-05-11:
Thanks for your report. Fixed bug is merged in the current master, next public release is expected in several days.