prusa3d/Prusa-Firmware #437: Random Layer Shift

tbedore, 2018-01-24:
I was just printing a regular file sliced on Slic3r PE, and there was just a random layer shift in the middle of the print.  I was there watching it happen, and there was just a layer shift randomly that happened almost as if it was part of the g-code.  It could be a problem with the firmware as I read people were having issues with the Mk3 and you said you were going to disable linear advance.  But also a problem with the slicer, I am running the most recent drivers and firmware.  The g-code that I printed is attached down below in the zip files.  
[Keychain.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/1661591/Keychain.gcode.zip)

Panayiotis-git, 2018-01-24:
The Linear advance feature can be turned off by using the "M900 K0" command. In Slic3r, switch to the filament settings tab and at the start g-code field, change the line with the M900 command.

Anyway, the uploaded g-code file doesn't contain any reference to the M900 command, so I'm not really understanding how the layer shift problem correlates with the linear advance feature.

tbedore, 2018-01-25:
Oh yeah, I heard that issue about the linear advance so I tried to disable it by just deleting the line of code completely.  But did you see anywhere in the gcode where a layer shift would happen?  I wanna say it happened around layer 2 - 4 it was still doing the bottom layers and the slicer is set for 4 bottom layers, but one layer was already created.

GurliGebis, 2018-01-25:
If I recall correctly, it remembers the K value between prints of you don't reset it or power cycle the printer.

sylphio, 2018-01-26:
It may be the same big layer shifting (+1mm) some forum user experience with firmware B150.
One guy on the forum say that disabling crash detection can fix the problem.
Personnally, i reverted back to B143 and this issues was fixed.

Reference with image : https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/layer-shifting-t13769.html

tbedore, 2018-01-26:
@sylphio The layer shift that I had is exactly like that, except this is running on the MK2s.

TheZeroBeast, 2018-05-30:
Referencing to PRUSA Printed Parts MK3 PullRequest I've just created addressing one cause of inaccurate Y-Axis homing which results in layer shift when a crash is detected.
https://github.com/prusa3d/Original-Prusa-i3/issues/76

To try you would install part, run Selftest in Firmware to set belt tension with new part.

3d-gussner, 2020-09-22:
@tbedore Is this issue still there with the newest firmware? If it has been foxed please close it.
[stale]

tbedore, 2020-09-22:
Not sure if it’s been fixed but I haven’t seen this issue in a long time so I’ll just close it

closed 2020-09-22 by tbedore