prusa3d/Prusa-Firmware #447: [Feature request - MK3] Make the printer PLA independent

szafran81, 2018-01-30:
Hi,

Since PLA is basically for kids (for toys etc.) it would be nice if the Wizard etc. asked to what temps (or what filament your using) to preheat the hot end.

CoolMatty, 2018-02-19:
The pre-heat menu offers a 215C temperature for PLA. That's a very safe number for all standard PLA, regardless of what the filament manufacturer might recommend for actual printing.

TimothyGold, 2018-02-19:
I would like to second this request.  I don't plan on buying any more PLA and will print primarily with PETG.  I already have been forced to re calibrate the XYZ and go through the first layer calibration.  It forces to use only PLA temps and does not get hot enough to calibrate accurately with PETG...  For now I have to make sure I have some of the Prusa Silver on hand....

szafran81, 2018-02-19:
@CoolMatty This has nothing to do with pre-heat temps

As @TimothyGold I don't use PLA at all (but I print in ABS or something with higher Tg). In about 5 years of printing I haven't used any PLA (mainly because it's just for toys, and only those that aren't used/stored in the sun). Well the first time I've printed any PLA was after putting together the MK3 kit. With this machine we're forced to buy PLA only for calibrating it. It's rather annoying.

3d-gussner, 2018-02-20:
@szafran81 You don't have to use the Wizard or LCD calibration, you can use your own calibration file with a gcode command in it to stop message "Distance ...not set...".
The Wizard and LCD calibration menu is ment to help beginners with the use and as everybody get's a spool of PLA with the Prusa MK2s/MK3 i think it is OK that Prusa 'just' has PLA in these two LCD functions.
You are a experienced user, so you will find a way around.

TimothyGold, 2018-02-20:
@3d-gussner I obviously missed something obvious and thanks for pointing this out.  Just this morning I realized that this was possible to adjust the Live Z offset and using this I realized I would just print a single layer test in any filament.  Thanks for the help.

szafran81, 2018-02-20:
@3d-gussner While I can do that, that's not the point. It should ask for the calibration temp (and propose one for PLA for newcommers) so one could calibrate using any filament he/she wants.
I don't remember the whole calibration process, but if you're saying that in the next few weeks if I'll rebuild my MK3 with the MMU upgrade and will be able to do the whole calibration without using PLA that's ok. If not then it's not.

3d-gussner, 2018-02-20:
@szafran81 It could be handy for some users but also confusing for beginners. Maybe a advanced user menu could help. On the over side the implementation of the Wizard and First layer cal. takes a bit of the limited ATmega resources. That's why the MK3 firmware has only 2 languages at this moment.

TimothyGold, 2018-02-23:
I would like to add one piece of information on this topic that I discovered.  If you ever redo a XYZ calibration is 0s out the first layer offset and every reboot I get a notice that the first layer calibration has to be redone.  So somehow it appears that this may be stuck in a loop until this flag gets cleared by going through the wizard that forces using PLA.  I still have some PLA so I was able to clear it but maybe someone else can confirm if this will be an ongoing issue for those of us that don't use PLA and maybe won't have any in the future.

workinghard, 2018-03-16:
All the  new printer are getting delivered with PLA roll. So it makes fully sense to keep all the default wizards to PLA. Even right now some people are already struggling with first layer, you don't want to make it more complex.
If you have any other material on hand only, just download the V2Calibration.gcode and modify the temps to whatever you want. Use it any time for calibration.

szafran81, 2018-03-17:
I don't get it - what's the problem ? All the things are allready there. Just add a question "What filament do you want to use for calibration?" and remember the selection. The temps are allready in the FW for 4 most popular filaments. It has nothing to do with people struggling with first layer. If someone is struggling then he/she would struggle doesn't matter what filament is used.

Someone said that there is no room in the FW - so make the FW single language, and make separate builds for different languages - and BOOM there you go a lot of new free space on the chip.

But anyways I don't care anymore. I've wrote my own version of the FLC g-code. Works like a charm. But it's still annoying that one needs a filament only for other printer calibration purposes.

If the people that take care of the code decide that it's a waste of time then this one is to be closed.

JohnnyDeer, 2021-01-13:
Actual printers and actual FWs are independent of PLA, as required in this FR, so I'm closing this issue.

closed 2021-01-13 by JohnnyDeer