prusa3d/Prusa-Firmware #449: Part cooling fan linear advance

kevinBecker, 2018-01-31:
Not a problem, but I believe there should be a linear advance setting for the part cooling fan as well. what if the part cooling fan could turn on slightly before the gcode says to ensure that the airflow is getting to the part on time. The fan takes a second to accelerate to its desired angular velocity.  From there, the air takes even more time to reach the nozzle through the fan-nozzle (because air is compressable).  This results in the beginning of bridges not getting enough cooling which leads to sagging bridges when printing in ABS (or where the fan is kept off and turned on for bridges).  By turning the fan on slightly earlier, bridges would look much cleaner

GurliGebis, 2018-01-31:
That makes good sense, though I think the only place it could be implemented would be in the slicer.
The problem is, if I recall correctly, that the "turn on" command is sent right before the part of the print that should be cooled is printed - in other words, the printer does not know it should turn on the fan untill then.

So it would require the slicer to know how long time it takes for the fan to spin up, and how fast all parts of the print is, so it can insert the command at the right place.
Also, it might be a problem is there is a big move (like a straight line) right before the part that require the fan to be on - starting the fan before the line might be too soon, but starting it after might be too late - so one might have to do half of the move, start the fan, and then the rest of the move.

I think it might be possible in theory, but getting it to work in reality might prove to be a problem.

Sebastianv650, 2018-01-31:
@GurliGebis is right, this has to be done inside Slicer. As Slice(3)r is already calculating the layer time it should be also possible to find the right point to insert the command.

neophyl, 2022-05-09:
I agree that the only place to do this is in the slicer, the machine firmware is not the best place for it.  Slicers such the the Super Slicer fork already have an option to do this.  It also has a fan kickstart value in case your fan needs a minimum speed to start up before dropping down to a low set point.

haarp, 2022-05-09:
> It also has a fan kickstart value in case your fan needs a minimum speed to start up before dropping down to a low set point.

That should definitely be in the firmware tho.

closed 2023-09-05 by github-actions[bot]