prusa3d/Prusa-Firmware #458: [Feature Request] - End of print beeper

mwarchut, 2018-02-03:
A lot of times I have many things to print and the printer is in another room.  It would be handy if it would beep when done.  Even if only a half dozen beeps.  And it should be configurable as an option to have it and how many times to beep.

servant74, 2018-02-03:
> <> ... Jack

(or send an email to my phone! <grin>)

-- 

The Four Boxes of Liberty - "There are four boxes to be used in the defense
of liberty: soap, ballot, jury and ammo. Please use in that order."
"Whatever you do, work at it with all your heart"... Colossians 3:23
"Anyone who has never made a mistake, has never tried anything new." -
Albert Einstein
"You don't manage people; you manage things. You lead people." - Admiral
Grace Hopper, USN
"The most dangerous phrase in the language is "We’ve always done it this
way"-- Admiral Grace Hopper, USN
"Tell me and I forget. Teach me and I remember. Involve me and I learn." -
Ben Franklin

Area5142, 2018-02-03:
As I remember, there is the M300 beep g-code command. You could put some in the end g-code to have the printer beep after a print. This way it would be optional to have the printer beeping...

dryja123, 2018-02-06:
I'll dig up the beep g-code when I get home.  A friend of mine lent me his S3D process and had an print start and print end beeps.  I had no idea and was across the room when my printer started to sound like a fire alarm....  Nearly had a heart attack until I figured out what it was.

vaxxi, 2018-02-07:
Just add whatever beeping you want in the end G-Code under your slicer's settings. It's:
`M300 S{pitch in hz} P{duration in ms}`

harrysamuel, 2019-02-21:
I found some code that almost worked for me. is there a way to have it beep till I press the control button?

SeanTomlinson30, 2019-03-29:
> Just add whatever beeping you want in the end G-Code under your slicer's settings. It's:
> `M300 S{pitch in hz} P{duration in ms}`

can you confirm what pitch you have set? Unsure what starting point I should/can use @vaxxi

cimoalpacino, 2019-03-29:
On fb groups you can find ending songs in gcode (look at Chris Warkocki's posts)

cimoalpacino, 2019-03-29:
https://www.ultimatesolver.com/en/midi2gcode

Eddiiie, 2019-04-10:
I have beep in my end gcode, here it is -

; Beep
M300 S0 P107
;M300 S2217 P160
M300 S1479 P321
M300 S0 P107
M300 S1479 P321

I made it from removing parts of a song until I had Beep Beep left.

bszonye, 2020-08-08:
I tried to use M300 to make an end beep, but the code does not seem to work on the new Prusa MINI yet.

kevinp2, 2021-05-07:
I will try the custom GCode but I would like to request this as a simple configuration item.  A good place might be:

Printer Settings > Advanced > General > Beep After Print Completed.

closed 2023-09-05 by github-actions[bot]