prusa3d/Prusa-Firmware #460: Hi Pitch noice coming from print fan

Richardgriff, 2018-02-04:
When the print cooling fan come on there is a high pitch noise. I believe this is cause by the high frequency PWM signal going to the FAN.

jettoblack, 2018-02-06:
On other Marlin-based printers I have fixed this by switching the part fan from hardware PWM to software PWM, which has a lower frequency and thus much less objectionable noise. I don't know if the Einsy has enough CPU power left over to run the fan on software PWM though, or if the Prusa FW still has the functions needed to run software PWM.

magkopian, 2018-04-06:
@jettoblack Increasing the PWM frequency so it will be outside the audible range can also get rid of the issue, without the need of using software PWM and loading the CPU even more. In Marlin this feature can be enabled by uncommenting the line `#define FAST_PWM_FAN` in [Configuration.h](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h). This has the side effect of the fan MOSFETs running a bit hotter due to switching losses, with most controller boards that shouldn't be a problem though.

vestom, 2018-04-28:
Just received and assembled my MK3. I noticed this distinct PWM whine when printing PETG, where the fan speed is "76". If tuning the speed to "255" (full speed) the noise disappears.

tdctaz, 2018-05-09:
This is really driving me insane, have a bunch of print I need to print in PETG and after 4 hours I had to abort the process, the constant whining is unbearable.

Rick-T, 2018-05-13:
+1

I have the same problem with my MK3 when printing PETG. The sound is really annoying.

justinvoelker, 2018-05-19:
I have nothing to add except a "me too" with the hope that more people means more attention from Prusa. I have completely stopped using the fan when printing with PETG and either completely avoid bridging or clean up nasty/fallen bridges afterward.

hagronnestad, 2018-05-24:
The whine is unbearable, I can hear it throughout my house. From the basement to the attic. Only when printing PETG.

mvasilakis, 2018-05-24:
I get this too. Sometimes its worse than others. It's definitely louder than it was when the printer was new. Matter of fact I get allot of odd noises that I didn't get when it was new.

maximlevitsky, 2018-05-24:
I have this too on my brand new MK3. Sadly in my humble opinion its just nature of PWM, I doubt that it can be fixed.

sorlok72, 2018-06-15:
New MK3 printer here arrived this week, same problem. with PETG it's really bad

sarf2k4, 2018-06-17:
Same here, glad I'm not the only one with this issues where the fan whines around 50% but gone at about 100%

Dolnor, 2018-06-17:
In the same boat here, would print ABS without a fan and PLA with layer fan at 100%, but switching to PETG I was amazed how loud this problem manifests compared to the general printer operation sound.

fedhat, 2018-06-22:
Same here. At first I thought it was an issue with the filament sensor or the pinda probe, but soon discovered it was the fan.  Voted up the issue in hopes that the firmware team can do something about it.

miksumortti, 2018-06-29:
This seems to work with no PWM noise, though probably far from the right way of doing it

`#define FAST_PWM_FAN` in Configuration.h

Then fix the scaling of fan speed

I just tested the unscaled fan operating range which seemed to be about 100-118 on my unit and botched a scaling to the M106 command processing in Marlin_main.cpp:

`fanSpeed=constrain(code_value(),0,255);`

changed to something like:
```
if (code_value() > 0)
    fanSpeed=constrain(code_value() * 18 / 255 + 100, 0, 255);
else
    fanSpeed=0;
```

asmaps, 2018-07-16:
I have the same issue when printing PETG. Really annoying!

Demo video: https://www.youtube.com/watch?v=Rf2mNBXZXNA

aemileski, 2018-07-19:
The noise is from power being applied to the fan out-of-sync with motor commutation, and from power being applied unevenly (PWM is "spikey").

Generally, intelligent fan control will cost additional CPU overhead.  Typically a dedicated micro is used for this.  [It is really easy to do with a tiny PIC12F1840, for example.]

Sync is easy to solve, as every motor commutation generates a tach pulse.  However, the tach signal should never be monitored while doing PWM, as there is a huge amount of hash on the signal.

Instead, to measure rotor RPM, for example:
1. Apply full power (100% duty).
2. Ignore the next tach pulse or two (usually very noisy, or incomplete).
3.  Wait for next tach pulse and start a timer (pulse is clean - can sync to it).
4.  Wait for next tach pulse and stop the timer.
5.  Restore the PWM control level.
6.  Calculate the rotor speed from elapsed time (two tach pulses per revolution).

The rotor will not appreciably speed-up at full power for only a revolution or two, because of rotor inertia and aerodynamic drag.  An internal PWM timer can be re-synced to the rotor, say every 100 revolutions (or more).

A good write-up with scope pictures is at http://www.pcbheaven.com/circuitpages/PWM_3_Wires_Fan_Controller_with_RPM_feedback/

The PWM power signal can be smoothed via dithering.  This technique is commonly used to prevent LED lights from blinking at low PWM duty-cycles, and valves from sticking during PWM dead-time.

Example:  50% duty cycle applies power over 8 intervals as 00001111.  That can be dithered into 01010101, same ratio of on and off periods, but much more consistent power application (dead-time spread-out).  Ideally a random distribution is used, rather than a repeating constant pattern, to further avoid any artifacts which might be audible.

A simple phase accumulator based (repeating pattern) dithering write-up of mine:  https://stackoverflow.com/a/35859455

OndraMac, 2018-07-24:
Same problem :( New fan on the way but its not the solution :(

plug-it-in, 2018-07-30:
One more for this very annoying noise , even 255 I have whining , Prusa please fix this

scorfman, 2018-08-02:
Same issue here. Replacement fan sent by PRUSA did not help.
Video https://drive.google.com/open?id=19QZF4HUR_tc4dW9Y62HQfnHJHeYSrEhl

0rsu, 2018-08-15:
Same issue here.  Kind of makes all the claims of noise reduction out to be a bit of false advertising.  Sounds like a fairly easy fix.  Please do so!

bamminger, 2018-08-15:
Same here with a new MK3. @Prusa, is it a fan issue and it has to be replaced?

bamminger, 2018-08-15:
Ok thanks, let's hope they fix it 😒

Zequez, 2018-08-15:
Are we sure it doesn't require a hardware change in the PCB though? I want
to imagine that if it were just a simple software fix it would be fixed by
now.

miksumortti, 2018-08-16:
^ I have been printing for couple months with [fast_pwm enabled](https://github.com/prusa3d/Prusa-Firmware/issues/460#issuecomment-401212225) without issues.

WannabeSpaceman, 2018-08-16:
@miksumortti noob questions, but do I need to append a value to the end of `#define FAST_PWM_FAN` after uncommenting it and how did you measure the unscalled fan speed to get to the values you did?

miksumortti, 2018-08-16:
@WannabeSpaceman Only uncomment. I uploaded the firmware to printer and changed the fan speed manually (settings->temperature->fan or M106 command through usb) to try at which value range the rpm of the fan changes.

Garpski, 2018-08-22:
Just in case Prusa are watching / taking notes here is a question for you. If its a software / firmware fix - can it be done or not? if it can be done why has Prusa not done it yet? and if it cant be done why haven't Prusa told us yet?

lenisko, 2018-08-27:
@miksumortti I have uncommented `#define FAST_PWM_FAN` in Configuration.h. How I'm supposed to find "unscaled fan operating range" on my Prusa MK3 ?

Was trying with your values in `Marlin_main.cpp` but got Fan Error. 
Fan itself was operating on high speed values, on smaller it was trying to start but couldn't or didn't reacted at all.

With only define change, fan was throwing Error or same symptoms as above.
This issue is really annoying..  I mean I can't even hear my printer EXCEPT those crazy whines from print fan :|

At the other hand, it's sad that Prusa team is ignoring it.

miksumortti, 2018-08-27:
@lenisko Low fan speed values will cause fan error due to the improper scaling method. Correct way would be to set the pwm timers frequency (?), so that the rpm calculating would also be right.
Also as the scaling is applied just to the gcode command, the printer settings fan speed input is not scaled.

Either increase the `MIN_PRINT_FAN_SPEED` to e.g. 106. (in Configuration_prusa.h), or disable fans check.

miksumortti, 2018-08-27:
@lenisko Tested XYZ calibration and it went through without errors.

lenisko, 2018-08-27:
@miksumortti thank you for checking. Probably something went wrong with compilation process on my side which is weird as I used recommended **1.6.9 Arudino IDE** and checked **v3.4.0-RC1** branch. Do you mind sharing own hex file?

acortellini, 2018-08-31:
I tried the @miksumortti solution, but the working range of the fan is really narrow, mine goes from 111 to 119.
I tried to uncomment the following line:

`#define FAN_SOFT_PWM`

and everything worked fine, the whining stopped and the working range of the fan is perfect.
To compile I followed the instruction on the main page, the source code used is ver. 3.3.1

acortellini, 2018-08-31:
If Someone want to give a try here the firmware with the modified line.

[Prusa-Firmware-3.3.1 -  Soft PWM.hex.zip](https://github.com/prusa3d/Prusa-Firmware/files/2340550/Prusa-Firmware-3.3.1.-.Soft.PWM.hex.zip)

miksumortti, 2018-08-31:
@acortellini Yep this works well and no need to waffle with fan ranges, just causes some ticking noise at the frequency of the pwm.

surrealseven, 2018-09-01:
I also noticed that my fan range became prohibitively narrow when enabling `FAST_PWM_FAN` out of the box.  I ended up setting the prescaler on the timer to 1024.  If my math is correct, that generates a frequency of about 60hz.  I can't hear the fan at all now, and the operating range seems to have even increased a smidge.

If someone else wants to give this a try, enable `FAST_PWM_FAN` in Configuration.h and change line 1054 in temperature.cpp:
// `setPwmFrequency(FAN_PIN, 1);`
`setPwmFrequency(FAN_PIN, 5);`

acortellini, 2018-09-01:
@surrealseven Good catch! The behavior is the same, but without the overhead of generating the PWM pulses. The fan range is very similar, the click in the fan is present in both solution, but I really don't mind, before I can hear the wining to rooms away...
I"ll go with your solution.
Thanks

haleypearse, 2018-09-07:
What about a low pass filter in series with the fan? This could smooth out the DC voltage. But I don't know whether adding a filter could damage the Einsy.

acortellini, 2018-09-07:
Sure you can install a filter, don't worry about the drivers, those chip are protected against everything. here the data sheet of the mosfets.
https://www.onsemi.com/pub/Collateral/NCV8402D-D.PDF

alpha1125, 2018-09-07:
This has been driving me nuts for a while... the whining fan, from the PWM...

@acortellini thank you for precompiling that firmware w/fix.

Now, if prusa will incorporate that...

alpha1125, 2018-09-07:
@haleypearse I don't see how a low-pass filter, such as a RC filter would damage the einsy in any way... but to be sure, one method could be put an opto isolator in between, thus shielding the einsey...

@acortellini did you use @surrealseven 's fix? or are you using your firmware? What's performing better?

lenisko, 2018-09-07:
I'm using @surrealseven proposed changes, everything cool and quiet from flashing time :-) clicking noise is not disturbing at all.
I'm leaving compiled __3.4.0-RC2, MK3 variant__ with fan changes only:
[Prusa-Firmware-3.4.0-RC2-fan-changes.zip](https://github.com/prusa3d/Prusa-Firmware/files/2362812/Prusa-Firmware-3.4.0-RC2-fan-changes.zip)

acortellini, 2018-09-08:
The two solutions ave the same effect on the fan, but I'm using the @surrealseven mod because the PWM signal is generated internally by the CPU, the soft PWM is using CPU power that can useful elsewhere. It's about a week end everything is fine printing mostly PETG.

alpha1125, 2018-09-08:
Note to anyone following along with this...

@acortellini's firmware, fails self test (at least it did for me).

@lenisko 's firmware, self test passes.  Thanks @lenisko ! 👍

acortellini, 2018-09-08:
I removed my firmware mod, please use @lenisko 's firmware.

trevjonez, 2018-09-10:
@lenisko do you I am not able to see a fork of the project under your account? can you post a diff of your changes?

lenisko, 2018-09-10:
@trevjonez Moved changes to github https://github.com/lenisko/Prusa-Firmware/commit/e48b4df815ed5ffca93dc06034b60cc039abb1f4

LeifSbr, 2018-09-13:
i've compiled the latest firmware (3.4 final) enabling FAST_PWM_FAN. fan noise is gone! BUT - performed the xyz calibration, noticed the z value in the display only shows a question mark (Z=?). hence i'm unable to do live z adjust as i have no idea what the current z value is - it stays stuck at "Z=?".
i'm using source code downloaded from the 3.4 firmware release page on github. i've since reflashed their precompiled hex and z value displays as normal. i'm compiling for mk2.5. any idea what i'm doing wrong?

Panayiotis-git, 2018-09-13:
Check the https://github.com/prusa3d/Prusa-Firmware/issues/1117

3d-gussner, 2018-09-13:
LeifSbr Is it a stock MK2.5 or do you have a mod with higher Z?

WilkoV, 2018-09-13:
My assembled MK3 makes the same noise and Prusa says it's the power unit.

trevjonez, 2018-09-13:
@WilkoV an easy way to test if it is the print fan is to go into the tune menu while a print in running and adjust the fan to 255 (which is 100%). if the noise stops. it is this problem

if you are running octoprint you can test it without starting a print by sending [`M106 75`](https://github.com/prusa3d/Prusa-Firmware/wiki/Supported-G-codes#m106) which is about 30% fan speed. then a `M106 255` to hit the 100% mark to see if it stops whining. then send M107 to turn it off.

LeifSbr, 2018-09-13:
@3d-gussner its a stock standard MK2.5.

@Panayiotis-git muchos gracias... that did the trick! i edited the correct platfrom.txt and all is well with the world - for now.

MrSahlholdt, 2018-09-15:
I just started printing in PETG today and now hear this noice from the fan. 
What is the solution so far? 
And is LeifSbr firmware still showing the "Z=?".

LeifSbr, 2018-09-15:
yup managed to get the noise down with custom firmware.
what i did was enable the line #define FAST_PWM_FAN in Configuration.h. ymmv though - not sure what or if there are any long-term effects? it seems to work ok so far...
also managed to resolve the "z=?" issue by editing the correct platform.txt file (see above post).

bsrotten, 2018-09-16:
Add me to another one fixed by this change.

plug-it-in, 2018-09-16:
I just put up with the horrible noise. The solution of changing the fan speed as a temporary work around worked, but I didn't want to install custom firmware, never tried a permanent change. I built a cabinet using Ikea tables and thats helped a lot.

alpha1125, 2018-09-21:
@lenisko any chance to compile the latest FW, and put it up again for the masses?

alpha1125, 2018-09-24:
@lenisko have you had a change to submitted a pull request with the changes you made? In case you don't know... see: https://help.github.com/articles/about-pull-requests/

I looked for a PR on prusa's branch... was this something you are comfortable with submitting as a fix? Or are there 'risks' that I may not be aware of.

Thank you for your time and effort on this matter... it makes it sane to be in the same room as the printer.

lenisko, 2018-09-24:
@alpha1125 those are originally @surrealseven changes. I would like to stay out of submitting PR to prusa3d/Prusa-Firmware - so far everything is fine with my own printer, but I don't have time to make research about those changes.

Feel free to ask @surrealseven or submit them yourself.

fjn777, 2018-10-08:
Hi
I use Ultimaker Cura sometimes and i tested to import the Prusa settings for Cura and noticed that fan settings are =100% and the extruder fan don't make that evil sound printing PETG. I haven't noticed any problems with my PETG parts yet.

[26 more replies]

closed 2020-09-21 by DRracer