prusa3d/Prusa-Firmware #461: Version 3.1.1.198 confirm "default settings loaded" during every startup

derjungewilde, 2018-02-05:
Hi,

I updated to latest version 3.1.1.198 from last weekend. Since than I have to confirm a message "default settings loaded" on the LCD screen every time. This is very annoying because I sometimes print remotely with octoprint and when I turn on my printer I'm not there to confirm this message. Without confirming octoprint will not establish a connection.

Affas, 2018-03-26:
Same problem here, any suggestions!

3d-gussner, 2018-03-26:
Newest MK2.5 and MK3 stable firmware is 3.1.3 and for the MK2 3.1.0.
Where did you get 3.1.1.198 from?

Affas, 2018-03-26:
I am using Version 1.3.6 Octoprint and latest Prusa update for MKS2. Never experienced this on earlier setup

3d-gussner, 2018-03-27:
@Affas So you have the issues with OctoPrint and the MK2s firmware 3.1.0 from 12th November 2017?
There is a M110 fix which has been pulled by Prusa on 20th December 2017 but Prusa didn't publish yet a new firmware release with it.

You should try to compile it yourself from the github source code
or try my custom firmware https://github.com/3d-gussner/Prusa-Firmware/releases look for the MK2/S/MMU firmware https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.1.0-001-HFM
If you choose my custom firmware PLEASE read the release notes!

robotrovsky, 2018-04-03:
Same problem here. I would like to know which default settings are meant? Because XYZ calibration is saved and definetly no default setting. So what should this message tell me? The default settings for which feature are loaded?

3d-gussner, 2018-04-04:
@robotrovsky Which printer and firmware version do you have?

robotrovsky, 2018-04-04:
@3d-gussner i have a prusa i3 mk2 with MMU and the latest official release 3.1.0

3d-gussner, 2018-04-04:
@robotrovsky Do you have that issue with SD or OctoPrint or both? If you use OctoPrint which version do you use? Latest official firmware release 3.1.0 from Nov 2017? If so read my comment above.

robotrovsky, 2018-04-04:
@3d-gussner Thank you for your support. Independent if i use OctoPrint or SD, whenever i turn on the printer i get the message. I read your comment and will consider to use your firmware. But actually i was only wondering to which setting the message refers. The default settings for which setting have been loaded? Do you know to which default setting the message refers?

3d-gussner, 2018-04-04:
@robotrovsky Check Marlin_main.cpp line 1247-> 1283

robotrovsky, 2018-04-04:
@3d-gussner Thank you very much! So it seems the reason is, that the version in the eeprom is wrong/old. I will try to save the settings with the M50X gcodes like described here:
https://github.com/MarlinFirmware/Marlin/wiki/EEPROM

robotrovsky, 2018-04-26:
BTW sending M502 followed by M500 like indicated in the EEPROM link removed "default settings loaded" message.

closed 2023-09-04 by github-actions[bot]