prusa3d/Prusa-Firmware #464: 3.1.1-201: Filament Sensor Not Responding Error After Update

dryja123, 2018-02-06:
Hello,

I updated my firmware to the latest version, 3.1.1-201 and I received an error "ERROR: Filament sensor not responding, please check connection"

The filament sensor worked perfectly just before I loaded the new firmware.  I used the auto load filament feature, ran a print, and updated the firmware after the print was completed.  

I resolve the issue by performing an all data clear and had to run the calibration wizard all over again.

mylife4aiurr, 2018-02-10:
Same thing happened to me. Filament Sensor working perfectly b4 update. Filament Status N/A right after update. I turned off printer from power switch and then back on, filament sensor seen by firmware. Probably needs a warning in download notes, so people can expect this and know how fix.

moojuiceuk, 2018-03-02:
Upgrading from 3.1.1 to 3.1.2 today and I had the same issue when the self-test runs. Filament sensor not responding. I had to Factory Reset - All Data then the self test responded correctly and recognised the filament sensor. A power cycle was not sufficient!

zafuquazar, 2018-03-02:
The same thing happened to me when I upgraded to 3.1.1 RC4. In communication with support I did a factory reset all data and that still did not get it running again. They have sent me a new sensor but I have not tried it yet. May not be related, but suspicious that it happens right at the upgrade.

mylife4aiurr, 2018-03-02:
Omg you guys are making me nervous. I just did upgrade for 3.1.1 to 3.1.2, its doing the self test as I write this. Finished - reports "Self test OK"
Printer firmware sees filament sensor. 

Im going turn on crash detection and give it another go, before it would occasionally hit nothing but the print, not anything else and detect a crash. So I disabled it.

dryja123, 2018-03-03:
Hey, OP checking in again.  Just updated the firmware to the latest version and got the same filament sensor error again.  Welp, guess I'm doing another factor reset.

omiks3, 2018-03-17:
I'm having exactly the same issue when updating to 3.1.2 and 3.1.3 tried factory reset and unplugging, both no good.
The only thing I could do to get it to work was to revert back to 3.1.1 and factory reset

BrendonW, 2018-05-28:
I need to follow this, my filament sensor has never worked. This gives me hope that it is a software problem, not hardware.

The replacement shows the same issue. I have the original sensor lying on the bench, but connected, next to the printer. The cell phone camera shows it is emitting IR light, but the firmware insists that it is not responding. I've tried two cables and two sensors.

If anybody has info on what to expect if I connect a digital scope to monitor the I2C lines, I can check if the firmware is talking to the board at all. I wish the error gave more info on the exact issue!

Postman64, 2018-06-04:
I wish I saw this thread yesterday before I started a long print. My sensor was working and I did a firmware update and only noticed the sensor was N/A when the nozzle clogged and the machine kept trying to print.
When this print finishes I’m trying a factory reset and if that won’t work I’ll load 3.1.2 and try that.