prusa3d/Prusa-Firmware #467: PINDA temp calibration - "Is steel sheet on heatbed? Yes/No" prompt displays twice then gets stuck

jettoblack, 2018-02-08:
MK3 with 3.1.1 release firmware

After going to Settings > Temp calibration, the LCD displays a message about needing a stable ambient temp, then it prompts "Is steel sheet on heatbed? Yes/No".  I select No. 

Next, the printer auto homes. Then the same message about needing stable ambient temperature displays a 2nd time, followed by "Is steel sheet on heatbed? Yes/No" a 2nd time.  I select No (again). 

Now the LCD remains stuck on "Is steel sheet on heatbed? Yes/No" for the duration of the procedure, which otherwise proceeds normally.  The LCD returns to the menu after temp calibration is fully finished.

This occurs whether or not Pronterface is connected via USB, but at least when it is connected, I can see the progress of the temp calibration via the serial messages.

TheZeroBeast, 2018-02-08:
Confirm that this happened when I attempted re-cal on 3.1.1

stahlfabrik, 2018-02-15:
Can confirm too! I read from someone in the forum who canceled temp calibration then because the machine looked hung.

Stiganator, 2018-02-25:
Can confirm this bug is present in 3.1.2 RC1 as well. It just displays "Is steel sheet on heatbed?" There should be a counter or something to let you know something is in progress, or at least a note that it may take X minutes to complete.

moojuiceuk, 2018-03-02:
I'm finding the same - 3.1.2 firmware and an "All Data" reset. Sat there for over an hour with a heated bed piping hot. About 100-105C with my laser thermometer.

Prusa-Support, 2022-05-05:
This issue will be closed due to it being obsolete. Please open a new issue if this is not solved yet

Marina Curzi
Prusa Research

closed 2022-05-05 by Prusa-Support