prusa3d/Prusa-Firmware #470: When continue from pause the heating of filament failed

loketak, 2018-02-08:
When I choose continue (Pokracovat) after I paused printing (Pozastavit tisk) the printer "stars heating" the extruder but in fact the current  temperature still fall down and set point temperature was set to 0. When I again in menu choose pause printing and try to heat it up from menu, the extruder correctly heats up. However after the extruder reach set point (for PLA 215 C) the set point was changet to 0. 
[image]

misan, 2018-02-11:
I reckon it is a duplicate of #221

GitHubMarine, 2018-02-27:
I'm having the same issue.  The nozzle fails to heat up and decreases in temperature instead of increasing after I selected "Resume" from a pause.

closed 2022-05-05 by Prusa-Support