prusa3d/Prusa-Firmware #474: [Request] Print from SD Confirmation

sarf2k4, 2018-02-12:
I would like to ask this confirmation to print feature because in some cases, when I was selecting the file to be printed let's say 5 files total, I wanted to pick file B, when I pressed that highlighted file, sometimes other files got selected to be printed which is file A or file C.

What to display on the print confirmation is the file name, if long it will scroll, and at the bottom, a "yes/no" selection

AnHardt, 2018-02-12:
That's not a solution for the problem. You'd have the exact same problem for selecting "yes/no".
The solution is to make the encoder less responsive.

sarf2k4, 2018-02-12:
It is about during the press of that encoder button where there's a chance that the cursor might jump in a fraction of a second before the button really got pressed.

AnHardt, 2018-02-12:
It's about turning the encoder while pressing it. It will not jump without reason.
Play with `ENCODER_PULSES_PER_STEP`.

sarf2k4, 2018-02-12:
I didn't turn the encoder while pressing it, No reason to do such thing and then trying to propose this =)

Panayiotis-git, 2018-02-12:
The clicks of the encoder do not match precisely the pulses it generates. There are cases when 2 clicks move the selection 1, 2 (mostly) or 3 rows in a menu.

This results to selecting a different file to print or load again the filament instead of opening the print menu or selecting to pause a print instead of the Tune menu, etc

macfly1202, 2018-02-12:
100% agree with sarf2k4. It's clearly not to the final user to deal with poor quality rotary encoder OR bad step library implementation.  Even if carefully doing the selection, it happens that it click on another line. On my MMU MK2S,  must reboot to step print and waiting the lonnnngggg sorting SD card. I've got an Arduino 2560 with a rotary encoder (ASM): this is clearly do a better job that on the prusa: I want to say more: it NEVER NEVER fails in click selection. Here the link to the rotary_encoder library in the project if dev's want to have a look:  https://github.com/cgwood/ArdustationMega.
So a confirmation is a great idea I think: thanks @sarf2k4

sarf2k4, 2018-02-12:
You are welcome, thank you on understanding and trying to justify what i
wanted to address as well.

Like i said in my opening post, to really confirm the print job because
when the printer starts heating the bed for abs took 5 minutes, we cant see
which file it is being printed until the printer starts.

An old zortrax m200 that has closed ecosystem and makerbot has the print
confirmation and also displays how long the job will take to finish.
Perhaps this also a good idea so each time before printing, user will know
that job will take how long to finish from that confirmation instead of
taking sd card to pc, open file, find for print duration.

=)

AbeFM, 2018-10-18:
This is a pretty common request - I've made and seen it made with Prusa and other manufacturers as well.

https://github.com/prusa3d/Prusa-Firmware/issues/855
This topic handles it much better and I think it should be considered in favor of this one.

sendingstrings, 2020-04-18:
I would like to see the chosen filename on the LCD after selection. 
Currently:
-select with encoder and click. (sometimes with an unintended encoder move at the last second.)
-LCD shows "Heating..........." on the 4th line. There is no feedback about what file was selected until much later. 

During the "Heating.........." phase, the [3rd line of the LCD is not displaying any useful information. ](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/printer-not-heating-correctly-temperature-problem-with-no-error/)

It would be really useful to have the filename selected displayed as a confirmation. Either on the 3rd or 4th line of the standard screen, or on an intermediate splash screen. The later could display even more info line filename, date, size, etc. 

That screen could also have a [click within x seconds] to return to the file list. 

Select file.
Click.
Glance at confirmation screen.
Walk away or click to go back.
5 seconds later the print starts.

AbeFM, 2020-04-18:
> That's not a solution for the problem. You'd have the exact same problem for selecting "yes/no".
> The solution is to make the encoder less responsive.

I disagree. Let's say there's a 20% chance of accidentally moving up, 20% of accidentally moving down, and the remaining 60 of getting it right.

Old way: (20% + 20%)  ->  40% chance of bad print.
New Way: (20% + 20% = 40% chance of wrong selection) * 20% chance (you can't go past "no", so you're 60% + 20% of being right) = 0.2 * 0.4 = 8%

That's a 5x improvement even if you do nothing else. I have a otherwise terrible printer with a bad touch screen that uses this confirmation method, and I've gotten 1 wrong print in months of use where I get several per week.

In practice, I find I can get a feel for when it won't work - sometimes it seems like it's very close to jumping - an encoded step and a click on the selector on not related, so it won't feel like you moved it, but you did. Adjusting those values you mentioned helps, I've done it on my own machines, but the improvement is nowhere near just making sure you're clicking on what you thought. You could logically debounce it, I guess, if the filename hasn't been highlighted for >1-2 seconds it won't select.... But that idea is blown out of the water by SS above:

> That screen could also have a [click within x seconds] to return to the file list.
> 
> Select file.
> Click.
> Glance at confirmation screen.
> Walk away or click to go back.
> 5 seconds later the print starts.

That's a great idea. I would be happy with that. What I don't like is coming back 15 minutes later to find my PLA printing at ASA temperatures messing up everything.

reddtoric, 2021-05-07:
Would really like if someone took a look at this problem and #855 . It's pretty cumbersome when I turn the knob to select something and it takes between 1 to 3 clicks to move cursor up or down one. If it's about the hardware/quality so it's hard to get it precise enough, then a confirmation of the print info should be provided (right away, not after some minutes it has started printing).

leptun, 2021-05-07:
In 3.10 the SD menu is much more responsive, so it will be much easier to select the right item.

AbeFM, 2021-05-17:
That's very good to hear.

Even though the brakes work great on my car, I still use the seatbelt. I'm
glad the menus are improved, it always seemed a performance issue not a
hardware one.

But it still remains that you can rotate the knob while you push it, and a
'Print File "blahblahblah.gcode"? Y/N' would help. :-)

I have this feature on a couple of my touch screen printers, it is MUCH
harder to mis select on those machines but I still appreciate the
verification. Even just immediately notifying the user of which file is
being printed would work, leaving it to them to stop the print manually.

closed 2023-09-05 by github-actions[bot]