prusa3d/Prusa-Firmware #476: [Feature request] extruder motor filament stuck detection.

zaico, 2018-02-15:
I would be nice if there was detection on the extruder motor, so if the filament got stuck outside the printer,                  eg. on the spool holder, it would sens it and pause the print and ask for help. 

I have had an episode that ended with the printer "pulling" the z axis crooked.

AnHardt, 2018-02-15:
Marlins focus is on software.
If there is a reliable, available, affordable filament motion device we will support it sooner or later.
But we will not develop such a device.

We can discuss pros and cons of concepts, cpips, ... here. Maybe you can win the heart of a single developer to make the hardware for a project here.

I know. It's a hen - egg problem.

AnHardt, 2018-02-15:
Just some random thoughts.

Would filament motion detection have helped with your accident?
No! The filament moved until it was too late. 

Are you looking for a filament spool motion detection system?
That motion is usually very uneven. Sometimes moving fast - than not moving for minutes. When not bowden only very loosely coupled to the extruder, more to head movement.

cnlson, 2018-02-25:
this should be part of the skipped step detection if the e motor is clicking when trying to feed from the spool