prusa3d/Prusa-Firmware #477: Files on SD card: The sort should be using the modified date

m-Th, 2018-02-15:
Hi,

The implementation of https://github.com/prusa3d/Prusa-Firmware/issues/169 isn't quite correct. 

The actual Sort by Date referenced in the above issue sorts the files using the Creation date but it must use the "Modified" date because we want to have at top the newest (latest modified) files.

Area5142, 2018-03-10:
That is why I get wired sorts when overwriting files on the AirFlash SD. Doing that a lot when designing new models.
I second that suggestion.

RobMeades, 2019-04-02:
Fully support this, see my thread [here](https://shop.prusa3d.com/forum/-f0/can-sd-card-sort-by-modification-date-not-creation-t28733.html#p135940), spent a day repeatedly printing a slightly modified file until I noticed the file at the top of the list wasn't my modified file!

draeath, 2019-04-02:
Hello,

The issue is the code uses creationDate and creationTime for sorting purposes (among others) in Firmware/cardreader.cpp - the original assignment can be found [here](https://github.com/prusa3d/Prusa-Firmware/blob/eea7554/Firmware/cardreader.cpp#L140-L141).

I don't understand the other usages of these variables in the file, but the uses within `CardReader::presort()` (definition starts [here](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/cardreader.cpp#L717)) could probably be adjusted to use modification times instead. You'd need to modify `CardReader::lsDive()` to supply those in addition to creation time, if you wanted to do this without changing those other references.

I do not yet have a printer of my own or I'd build and test a patch for a PR, but hopefully this helps someone else do so!

draeath, 2019-04-16:
PR #1739 submitted - this fixes it on my MK3S.

draeath, 2019-09-06:
Well, this has been sitting unreviewed since I submitted the PR. I'm sorry :(

Wish I knew how to get eyes on this.

RobMeades, 2019-09-09:
I suspect it's not seen as a high priority as the default file name out of Prusa Slicer has changed to include the expected duration of the print and hence the file name is very likely indeed to change between iterations of the same job.

Of course this fix IS still required since there are those of us who prefer to stick with the same file name rather than fill our SD cards with a gazillion things that have slightly different names where the difference in the name is not visible off the left of the LCD display anyway!

Panayiotis-git, 2019-09-09:
Very-very useful!

draeath, 2019-09-10:
PR tagged for firmware 3.9.x!

commit e56cb14b (2019-11-04): Merge pull request #1739 from draeath/MK3-fix-sdcard-sorting

use modification times for sdcard time sorting. fixes #477
closed 2019-11-04 by DRracer