prusa3d/Prusa-Firmware #478: Option to turn off the beep on i3 MK3

Simahnyk, 2018-02-17:
Hello all, who would read this post!
I would really appreciate the option to turn the beeping off on i3 MK3. The reason is simple - the printer I'm using is owned by the grammar school, where I teach. It is being used in the space adjacent to the physics classroom. And here the problem comes - the printer is separated from the classroom by only a thin wooden wall (looks like plywood or MDF board, just painted over). When I set up a print with a color change or two on certain layers, I know I won't be able to visit the printer sooner than in, say, 50 minutes. When the color change comes 5 minutes after I leave the room, the printer will beep for full 45 minutes. And that is a big issue when there are students having a physics lesson behind the very thin wall... the beeper is very loud, annoys students and the teacher behind the wall, and practically can't change a thing, because I, being the only operator of the printer so far, am teaching on the lower floor, so I not only can't hear the beeper, but I couldn't leave the lesson even if I heard it.

I noticed that Prusa made a very wise decision to implement automatic cooling of everything when the printer is waiting for a filament change, so when I come after some time the printer is cold and waiting for me to change the filament to another color, it has to preheat the nozzle before I can perform a change. This is cool, because I really can't come sooner even if I knew the time for filament change has come, and I'm really glad that I don't have to worry about the filament being baked in the nozzle for an hour before I come. BUT... the beeping is a great problem. Since 3D printing is a time consuming process, I absolutely have to use the printer during lessons in the adjacent classroom. And the beeping disturbs them very much. Maybe even to the point, where I'll have to stop using the printer... which would be very sad... or where I'll have to open the case and make a tiny modification to the piezo beeper itself (like shoving a screw into the hole in the piezo...).

But this is a change that should be extremely simple to program into the firmware and I believe it would help not only me, but many people, who use the printer in the close proximity of other people, who are not familiar with the printer, and who can't come to change the filament right away. I recognize the necessity of alerting the operator when something is wrong or needed, but this can be done by, say, 10-15 seconds of beeping (which is very loud indeed). If I'm nearby, I know the time has come and I do what must be done; if I'm away, the beeping won't disturb anybody if it lasts only for 15 seconds.

So... I'd like to see a setup menu option to set the length of beeping between "indefinite" and "15 s" for example (maybe even "off" completely), with "indefinite" preset from factory. What do you think, would this be good for anybody else?

Thank you and keep up the good work! :-)
Satisfied user of i3 MK3,
Michal Hnyk

jreszka, 2018-02-28:
Second that! Now i cant print because I need to change filament but my kids just fall asleep. If I beep, I'm dead man.

zafuquazar, 2018-03-01:
Yes please!

Area5142, 2018-03-10:
Beeping should stop after a small count or a time limit.
I have a microwave oven that keeps beeping - it is very annoying when things do that...

Tyrandor, 2018-03-13:
+1 - need an option for turning it off completely or after a time limit. I cant run the printer overnight if the filmanet sensor could fail again and waking up the whole house with the beeping terror.

polymetric1, 2018-03-17:
Yes, this is an absolutely horrible sound. Add the option to get rid of it in the settings menu, and when it's on normally it should also probably be at least 75% of the volume it's at currently, as well as being a lot lower pitch.

jhoff80, 2018-03-19:
I'm on the MK2.5 but I agree in full.

mbelchuke, 2018-04-16:
No fix for this yet? I live in an apartment, and a beeping printer is really annoying for my neighbors, if a print fails while I'm on work.

koryba, 2018-04-19:
same for me. Please consider also the beeping on filament change

raster, 2018-04-24:
Yes, please! We're in a shared environment and are getting complaints (and threats) about the beeping.

jettoblack, 2018-04-27:
Yes please! You can call it "Disable wife smashing your beeping printer" option. :)

polymetric1, 2018-05-05:
Prusa seems to be completely ignoring this issue. This is insane, it's an incredibly simple fix for someone who knows this firmware well, and basically impossible otherwise, even if you're somewhat familiar with the marlin codebase. They seem to have completely overhauled the default Marlin configuration files, at least. There is absolutely no debate to be made whether something should be done about this. You can't market a printer for being silent and then have it beep at like 70dB whenever one of its main features is used.

aequitas, 2018-05-05:
@polymetric1 if you want to complain go to Prusa helpdesk or something, but don't pollute Github issue threads with useless comments like this. If it is as easy as you say just make the change yourself and build your own code, or better yet make a contribution with a PR :)

coreyward, 2018-05-21:
The beep is introduced by the slicer using gcode `M300`. You can do a find/replace on your gcode file to remove any `M300` line, or you can recompile and flash your machine with [these lines](https://github.com/prusa3d/Prusa-Firmware/blob/4ce950c/Firmware/Marlin_main.cpp#L3496-L3518) adjusted: 

```cpp
// Line 3496-3518
case 300: // M300
    {
      int beepS = code_seen('S') ? code_value() : 110;
      int beepP = code_seen('P') ? code_value() : 1000;
      if (beepS > 0)
      {
        #if BEEPER > 0
          tone(BEEPER, beepS);
          delay(beepP);
          noTone(BEEPER);
        #elif defined(ULTRALCD)
		  lcd_buzz(beepS, beepP);
		#elif defined(LCD_USE_I2C_BUZZER)
		  lcd_buzz(beepP, beepS);
        #endif
      }
      else
      {
        delay(beepP);
      }
    }
    break;
    #endif // M300
```

Replace with this: 

```cpp
case 300: // M300
    {
      // Don't play beeps
    }
    break;
    #endif // M300
```

I'm not familiar with how options/settings are configured and written, but it would seem like a good solution would be to add a “Disable Beeps [on/off]” to the Settings submenu. @aequitas If you can maybe point me in the right direction I can build something out.

jelson, 2018-05-21:
@coreyward That's not accurate - while the Prusa firmware does support beeping in response to gcode commands, the code at issue for this bug is [here](https://github.com/prusa3d/Prusa-Firmware/blob/4ce950c/Firmware/Marlin_main.cpp#L1514) - the beeps that are generated after filament runout is detected until the user clicks the LCD button. The firmware can be recompiled without BEEPER defined to turn off the beep.

(The Slicer can't know when the filament has run out, so its beep generation can't be what's causing the beeping people on this bug are complaining about.)

coreyward, 2018-05-21:
@jelson Ahh you're right… and it looks like the beeper code due to filament run-out is actually in a couple places, and it has the same fallback to use the LCD buzz.  

Is this file (`Marlin_main.cpp`) the source or is this being generated by a build script somewhere?

3d-gussner, 2018-06-07:
@modanielac the  "//#define FILAMENT_RUNOUT_SUPPORT" isn't used for the new filament sensor that is pat9125 , the FRS is used for a mechanical or optical switch which can be connected to a pin. Used that for the MK2/s.

modanielac, 2018-06-07:
Ah, in which case, I have no idea which lines deal with the buzzer in the case of runout!

modanielac, 2018-06-08:
Ok,
 I now know exactly how it works. the state of a filament sensor is read by frsensor and issues a M600 command if the filament is not detected correctly. So it is the buzzer/beeper settings in Marlin_main.cpp under "case 600" which need to be changed.

modanielac, 2018-06-08:
done:

 the code

		uint8_t cnt = 0;
		int counterBeep = 0;	
		**int counterBeepIterations=40;**
		
		fanSpeed = 0;
		unsigned long waiting_start_time = millis();
		uint8_t wait_for_user_state = 0;
		lcd_display_message_fullscreen_P(_T(MSG_PRESS_TO_UNLOAD));
		while (!(wait_for_user_state == 0 && lcd_clicked())){

			//cnt++;
			manage_heater();
			manage_inactivity(true);

			/*#ifdef SNMM
			target[E_AXIS] += 0.002;
			plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], 500, active_extruder);

			#endif // SNMM*/

			//if (cnt == 0)
			{
#if BEEPER > 0
				**if (counterBeep == 500 && counterBeepIterations < 10) {**
					counterBeep = 0;
					**counterBeepIterations++;**
				**else{
					WRITE(BEEPER, LOW);
					counterBeep = 1;**						
				}
				SET_OUTPUT(BEEPER);
				if (counterBeep == 0) {
					WRITE(BEEPER, HIGH);
				}
				if (counterBeep == 20) {
					WRITE(BEEPER, LOW);
				}
				
				counterBeep++;				
#else

modanielac, 2018-06-08:
int counterBeep = 0;	
		int counterBeepIterations=40;
		
		fanSpeed = 0;
		unsigned long waiting_start_time = millis();
		uint8_t wait_for_user_state = 0;
		lcd_display_message_fullscreen_P(_T(MSG_PRESS_TO_UNLOAD));
		while (!(wait_for_user_state == 0 && lcd_clicked())){

			//cnt++;
			manage_heater();
			manage_inactivity(true);

			/*#ifdef SNMM
			target[E_AXIS] += 0.002;
			plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], 500, active_extruder);

			#endif // SNMM*/

			//if (cnt == 0)
			{
#if BEEPER > 0
				if (counterBeep == 500 && counterBeepIterations < 10) {
					counterBeep = 0;
					counterBeepIterations=counterBeepIterations+1;
				else{
					WRITE(BEEPER, LOW);
					counterBeep = 1;						
				}
				SET_OUTPUT(BEEPER);
				if (counterBeep == 0) {
					WRITE(BEEPER, HIGH);
				}
				if (counterBeep == 20) {
					WRITE(BEEPER, LOW);
				}
				
				counterBeep++;				
#else

please can someone try it ( I would but for warranty issues)

modanielac, 2018-06-08:
I have the modified code.. but I don't know how to include a code snipet here.
I am having a problem with formatting.

modanielac, 2018-06-08:
so I have attached a snipet.
 please can someone try it.

 I have included the modified code as a seperate file:

[modifiedcodesnipet.txt](https://github.com/prusa3d/Prusa-Firmware/files/2084605/modifiedcodesnipet.txt)

And the complete modified file:
[Marlin_main.txt](https://github.com/prusa3d/Prusa-Firmware/files/2084609/Marlin_main.txt)

kbx0, 2018-07-10:
+1 for this feature please. I'd really like to see an official function.

BigAl66, 2018-07-25:
+1
The printer is so quiet while printing. I also hate the loud beep during filament change. I already thought about using my solderin iron...

halfcab1981, 2018-07-31:
+1 for this feature. I have been told not to print at night by the wife because of the beeping.

Bulli, 2018-08-01:
+1 I use the printer in my office, and the beeping annoys my colleagues.

3d-gussner, 2018-08-01:
If you check the pull requests it looks like more sound/beep option is in development.

progman32, 2019-06-14:
Looks like the feature got backported from MK3's firmware! https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.2.3

mavelli, 2019-08-01:
mk3s with 3.7.2-2363 here - in the settings, i can mute the beeper for regular operations, but when i turn off the power (on the power supply) the machine is beeping super-loud and long before turning off completely. is this also in developement to be customizable? if not, i am thinking about adding a resistor to the beeper or just removing it alltogether.

GurliGebis, 2019-08-01:
It is due to the power panic logic being triggered, which is a bug.
It has been fixed and the fix will be in the 3.8 release.

mavelli, 2019-08-03:
power panic logic trigger bug fix - awesome! hope the 3.8 is releasing soon.

leptun, 2020-01-26:
There is an option in the settings/tune menu to set sound. Use `Once` or `Silent` and it should stop all beeping (with the exception of serious errors).

fscheps, 2020-08-30:
It would be cool if instead of "Super loud" or "Completely Silent" we would still have the option of an audible noise, but that doesnt wake up my son when sleeping in the room next to the room where I have the printer. Thank you for your consideration.

greenveg, 2022-03-03:
+1 for more option on beeping sounds.

My main problem is of the out-of-filament-in-the-middle-of-the-night kind.

Adding another button press before extruder heating starts would help a bit. Then at least my partner can just press the button and not be disturbed when working from home if it runs out of filament.

DBrendebach, 2022-03-17:
I had a similar problem and found this:

`Mute beeper when prompting color change

If the estimated color change time happens to be during the night or another inconvenient moment, you can switch your printer’s beeper sound to [Silent], [Mute] or [once] mode. It is found under Settings, on the LCD-menu`

ascheucher, 2022-05-16:
Still the press once to mute, press a second time to continue would be great!
I ran out of filament and it's beeping now since 24 hours.
New filament is ordered, but it will be another 48 hours of beeping till it will arrive and the beep will finally stop and all our nerves relieved.
I really don't want to through away 32 hours of printing. But it's a hard time...

tarheele, 2023-04-23:
> Yes please! You can call it "Disable wife smashing your beeping printer" option. :)

100%

matthenning, 2023-04-27:
I've gone ahead and disconnected my display and only use Octoprint, as false thermal warnings kept waking everyone up.
Please let us mute this speaker already!

Prusa-Support, 2023-06-29:
The main issue in the original post should be already been solved back in September 2019 with the introduction of _silent sound mode_ in [FW 3.8.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.8.0).

Please use _silent sound mode_ so that beep sounds will only be used when strictly necessary.
The option to definitely turn off the sound - even for major warning events that basically mean "the printer should _not_ be used due to hazardous conditions" - is intentionally not allowed for safety reasons as you can read in the same [FW 3.8.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.8.0) release notes.

About volume adjustments, this is not allowed due to hardware limitations as explained in the specific issue
https://github.com/prusa3d/Prusa-Firmware/issues/3444.

At this point, I wonder if this issue can be closed.

Michele Moramarco
Prusa Research

closed 2023-06-29 by Simahnyk