prusa3d/Prusa-Firmware #484: Err: Front fan error when using a model with 0 bottom layers

Paulbsh, 2018-02-20:
I just noticed that I get a front fan error when I'm trying to print a model with bottom layers set to 0.
This happens even when the fan isn't even supposed to run, i.e. if cooling is set to start at layer 3, the fan error occurs on layer 2.
Slic3r, Slic3r PE (Prusa PETG preset) and S3D was used for testing, error every time.
As soon as I add 1 bottom layer, there is no error.
Recent Firmware 3.1.1

bubnikv, 2018-02-21:
Please attach your G-code that causes the print error issues. Thank you.

Paulbsh, 2018-02-21:
[cube.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/1744011/cube.gcode.zip)

bubnikv, 2018-02-21:
Thank you for the data, we will look into it.

PavelSindler, 2018-02-22:
PR https://github.com/prusa3d/Prusa-Firmware/pull/492 fixes this issue. Thank you for cooperation.

closed 2018-02-23 by PavelSindler