prusa3d/Prusa-Firmware #498: Statistics issue in 3.1.1

cnlson, 2018-02-23:
This will take reviewing the pictures while i narrate. After a print ends the time continues to accumulate on the screen, as if the print is continuing. Once you start a new print the accumulated time from the last print is added to the statistics. So the time is basically the entire time you have had the printer on since the last time off _unless_ you reset the printer or turn the printer off. 

in this picture note the accumulated time
[image]
in this picture note the print time of 00:25 minutes and the apparent accumulating time since the print of 11 hr 17 min
[image]
starting a print resets the clock 
[image]
and appends the 11;17 onto the total time
[image]

resetting or turning off and on completely stop the counter accumulation
[image]

magkopian, 2018-04-04:
I have the same issue and I'm running the latest stable firmware release 3.1.3. I just finished printing a calibration cube on my new MK3 and that was the result. So far it happened only once, but I haven't done more than 15 prints in total.

[image]

tofb15, 2019-08-15:
Just encountered this on my MK3s with MMU2s running firmware 3.8.0.
[image]
[image]
[image]

Each print adds 1000+ hours to the Statistics.

closed 2023-09-05 by github-actions[bot]